FreeCAD/FreeCAD-Bundle #247: topological naming issue with 0.22.0dev

dewey86, 2024-06-12:
Hello Guys.

After watching a youtube video which praised "freecad 1.0" for solving the topo-naming-issue, i downloaded and installed it.
I immediately created a fairly simple geometry to test if the topo-naming-issue is really solved.
Unfortunatly not in this case.

When Changing the "Pocket" from 5mm depth to 10mm or "through all", the "Fillet" from the left hole switches to the bottom of the second hole.
Before changing the depth of the "Pocket":
[image]
After changing the depth of the "Pocket":
[image]

Here is the model:
[FC-Test_.20240612-085512.zip](https://github.com/user-attachments/files/15800987/FC-Test_.20240612-085512.zip)

shaise, 2024-06-12:
Quite similar to what I encountered.

casterle, 2024-06-12:
From what I've read, I believe that the TNP fix is not yet included the publicly available builds. It should be available soon.

dewey86, 2024-06-13:
> From what I've read, I believe that the TNP fix is not yet included the publicly available builds. It should be available soon.

Hmm, okay. In this video they claim to be solved: [https://www.youtube.com/watch?v=7-GEKBQPpyY&themeRefresh=1]

The waiting seems to contiue :)

casterle, 2024-06-13:
I only speak English, so didn’t understand the video. Here’s one that addresses the subject in English: https://www.youtube.com/watch?v=tYNkZ-ry5yA

 

The video demonstrates the fix, which will be included in version 1.0 which should be released in mid-July. It may be included in current dev builds, but as I haven’t checked I don’t know for sure.

I found this on MangoJelly’s Patreon page:

From the dev meeting today.

Feature freeze: monday 2024-06-03
UI freeze: monday 2024-06-24

Estimated release time: mid-July (2024-07-15)

FreeWilly2022, 2024-06-15:
The TNP is being mitigated, not solved. As one of the commenter's put it, they can break any current 3D program like this. I have TNP issues with Fusion 360. It seems to be endemic to this type of CAD where existing features are referenced. Anyway, I tried both this bundle and compiling FreeCAD from the current source, and both appear to have the TNP mitigation implemented. However both are suffering from some serious problems where simple changes cause the model to self destruct. I added some rows to the spreadsheet, not changing anything in the model, and my model broke. When it broke there were lengthy reports of renaming of edges, so it seems to be a problem with the TNP solution. Looking through the Issues and Actions on the main FreeCAD github it looks like there are a bunch of fixes that the devs are working on. So I think am going to wait until the devs make more progress before trying again.
One strange difference I don't understand is that when I compile from the source code there are substantial improvements/changes to the UX, particularly the icons in the sketcher, that I don't see in this bundle. Seems like more than just a style sheet difference???

casterle, 2024-06-16:
>  I added some rows to the spreadsheet, not changing anything in the model, and my model broke.

I use and mostly like the spreadsheet. I've learned to be careful with it, never inserting rows in lookup tables for example. FreeCAD's not perfect, but since I started using it three or four years ago it's improved by leaps and bounds. As I understand it, the Foundation (to which we all should contribute a few bucks each month) has hired full-time programmers. And Ondsel has been contributing their improvements as well. 

I don't claim FreeCAD is perfect or that it's the right solution for everyone, but I get sick of hearing folks dismiss it out of hand based on past experiences or the word of others. It's plenty good enough for many of us, and I'm willing to put up with its quirks and bugs in return for the low, low price of admission.

I came to FreeCAD from the community version of F360, at the time because . I added some rows to the spreadsheet, not changing anything in the model, and my model broke.

FreeWilly2022, 2024-06-16:
@casterle - sorry I was not very clear in my comment. What I mean is that we who use the dev version are supposed to post issues that we find to help the dev's. But I think the dev's are working on enough issues right now that I am going to wait a few days until they get the fixes they are currently working on done. Then I will see if the issue I found is still a problem before I post an issue.

maxwxyz, 2024-07-04:
Is this still an issue in the latest weekly?
If so please create an issue (if there is not already an existing one) here https://github.com/FreeCAD/FreeCAD/issues

closed 2024-07-04 by maxwxyz

shaise, 2024-07-05:
I can confirm this is still an issue. I reconstructed the model from scratch just to be sure the issue is from some bad things in the attached model.
I can also confirm that on Realthunder's LS3 version, everything works perfect.
@dewey86 , please open an issue as suggested above.