FreeCAD/FreeCAD-Bundle #248: Threaded hole not working in Part Design

buteomont, 2024-06-13:
Something has broken about **threaded holes** in the _Part Design_ workbench.

I started a new project, added a body, created a simple **40x40mm** square sketch, and padded it to **40mm**.

Then I selected the top face, pressed the _Hole_ button, and chose the _ISO metric fine_ profile. I selected the _Threaded_, _Model Thread_, and _Update view_ checkboxes. 

I set _Size_ to **M22x1.0** and _Drill point_ to **Flat** and pressed the _OK_ button.

The hole never showed up on the model, but it does show in the feature tree.

I'm using development version **0.22.0dev** (FreeCAD_weekly-builds-37668-conda-Linux-x86_64-py311.AppImage) on Ubuntu 20.04.  This problem does not occur in version 0.21.2 (FreeCAD-0.21.2-Linux-x86_64.AppImage)

FWIW, I'm just learning FreeCAD and the more I use it, the more I like it and appreciate the power within. Kudos to the creators!

edragos, 2024-06-13:
You can not create a Hole from Pad in any version of Freecad.
You must select a face, create a new sketch, draw a circle, position the circle, close the sketch.
Select this sketch and make what hole you desired.

buteomont, 2024-06-13:
My apologies, I think a lack of sleep has muddled my brain. A problem exists, however. I have updated the original post with the correct steps to show the problem.  Basically, the hole shows up but not the threads, and an error is generated.

edragos, 2024-06-13:
No problem. You’re welcome.
Also, I am new in Freecad, I learn every day something new.
Let me have a look at this problem.
My gratitude goes to this marvelous community around the globe, that work very hard to make this wonderful software and keep it free.

buteomont, 2024-06-16:
A clue:  I discovered that if I modify the hole to pass completely through the cube (by adjusting the depth or selecting _Through all_) the problem goes away. 

As a side note, selecting _Through all_ sets the hole depth to 139.95 mm for some reason, even though the cube is only 40 mm thick.

buteomont, 2024-06-17:
Another clue: It seems that the problem also disappears when I set the hole's **Refine** value to _false_.