FreeCAD/FreeCAD-Bundle #250: Weekly builds file naming no longer following legend.

TheCakeIsNaOH, 2024-06-14:
The legend indicates that the weekly builds include a date in the filename:
[image]

However, recent builds do not include a date:
[image]

Is this an intentional change, or an oversight in the release process? This filename is used to generate the version for the Chocolatey package, so format changes break the automation.

maxwxyz, 2024-07-04:
@adrianinsaval FYI

adrianinsaval, 2024-07-04:
The change was made on purpose because the date of the upload can now be seen on github, so it gave no useful information just making it unnecessarily long. I'll try my best to keep the format stable from now on.

adrianinsaval, 2024-07-04:
Since it looks like they've fixed it on the Chocolatey side I will close this issue as changing it back will just create problems again

closed 2024-07-04 by adrianinsaval