FreeCAD/FreeCAD-Bundle #252: file writing problem

yoyo325, 2024-06-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Try to install Freecad 02.1.2 and received the following error:

error writing file
c:\Programm Files\Freecad02\bin\Coin4.dll

Is this again a mining programm?

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * 3DfindIT 1.2.0
  * Assembly4 0.50.2
  * fasteners 0.4.56
  * freecad.gears 1.0.0
  * parts_library
  * pcb
  * ThreadProfile 1.85.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-04:
@adrianinsaval FYI

adrianinsaval, 2024-07-04:
Sounds like an overzealous antivirus or corrupted download. Not much we can do on our side

yoyo325, 2024-07-05:
Hi,

Meanwhile problem was solved, don’t really know what happened during download, a few days later download succeeded 

Anyhow thanks for support.

Regards

Yoyo325

closed 2024-07-05 by maxwxyz