FreeCAD/FreeCAD-Bundle #253: Body colour keeps reverting to grey

trilbytim, 2024-06-15:
**Summary**
Body colours are not being preserved in latest Weekly Builds dev version

**Steps to reproduce**

- Create a body
- Set the body colour to anything other than the standard grey in View> Shape Appearance
- Draw a sketch
- Do a pad
- Body colour will be set back to standard grey, so set it back to the colour you want
- Do a fillet
- Body colour will be reset to grey

In other words, it seems that literally any solid body operation is resetting the colour back to grey. Also the 'random colour' option seems to randomly choose grey every time!

**Version**
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git) AppImage
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CfdOF 1.25.13
  * OpenTheme 2024.5.3
  * Plot 2022.4.17

shaise, 2024-06-15:
Same in sheetmetal. I guess it has to do with the new material system.

maxwxyz, 2024-07-04:
Is this still an issue in the latest weekly?
If so please create an issue here https://github.com/FreeCAD/FreeCAD/issues

closed 2024-07-04 by maxwxyz