FreeCAD/FreeCAD-Bundle #256: Thema can not be applied

borish43b, 2024-06-19:
I am not able to change the theme.  

**Steps:**

1. Select "Edit -> Preferences..."
2. Change "Theme"
3. Click "Apply" or "OK"

**Expected behavior:**
The theme is changes to the selected one

**Observed behavior:**
The following error is yielded:
`boost::filesystem::create_directories: Zugriff verweigert [system:5]: "C:\Users\<USERNAME>\AppData\Roaming\FreeCAD\SavedPreferencePacks\Backups", "C:\Users\<USERNAME>\AppData\Roaming\FreeCAD"`

**Additions:**
- The selected theme and all other settings are persisted. If I close and restart FreeCAD, the selected theme is shown in the preferences. The theme is not applied, however.
- All other settings (e.g. language) take effect as expected.
- My name contains "german umlaut" letter (ä). There has been some issues with this in the past already. Maybe this causes the issue.

**Build info:**
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: German/Germany (de_DE) ]

borish43b, 2024-06-22:
This is definitely an issue with the character encoding in the path (again, there had been an issues with this for the plugins a few versions back).
I have created a new user “Tester” and everything worked fine.
Then I have created another user “Täster” and I got the exception again.

Also applies to the newest build.

maxwxyz, 2024-07-04:
Closing this for the issue in the FreeCAD repo

closed 2024-07-04 by maxwxyz