FreeCAD/FreeCAD-Bundle #260: Dependency graph view broken (regression)

krushia, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Dependency graph view is broken. Export still works though.

Last working build:
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2

First broken build:
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git) AppImage
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2

[image]

### Full version info

```shell
^
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-07-05:
on windows it's even worse as not even `dot.exe` was being shipped. This should get fixed now: https://github.com/FreeCAD/FreeCAD-Bundle/commit/81ed77414f7da7c3884acfae88ff459473fa0fe2

adrianinsaval, 2024-07-05:
I created a seaprate issue for the default search path: https://github.com/FreeCAD/FreeCAD/issues/15224

closed 2024-07-05 by adrianinsaval