FreeCAD/FreeCAD-Bundle #261: Sketcher: Sketch001: Argument not Found

FeedTheChunk, 2024-07-05:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.14
  * fasteners 0.5.24

After building a sketch and pading I create a sketch on one of the faces.  When I attempt to import geometry "Sketchoo1: argument not found." error.

I've tried this one two different models and both trow the same error message.

[image]

JohnOCFII, 2024-07-06:
incorrect repo for this report.  I have also seen this issue with 38001.  Might already be fixed.  Correct place for this issue is: https://github.com/FreeCAD/FreeCAD/issues

adrianinsaval, 2024-07-06:
see https://github.com/FreeCAD/FreeCAD/issues/15117 a build with the fix should be up soon. and as mentioned this is not the repo to report these kind of problems, this one is exclusively about packaging.

closed 2024-07-06 by adrianinsaval