FreeCAD/FreeCAD-Bundle #262: Sketcher add dot crashes FreeCAD_weekly-builds-38043-conda-Linux-x86_64-py311.AppImage

jnweiger, 2024-07-06:
Seen with https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-38043-conda-Linux-x86_64-py311.AppImage

* In Part Design, I click on the sketch icon. Sketcher opens. choose e.g. XY-plane
* Select red dot icon Create Point
* place the point anywhere. 
* Freecad crashes

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7fd0da155090]
#1  0x7fd0dbf5fead in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADApp.so+0xed
#2  0x7fd0dbf60c97 in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADApp.so+0x297
#3  0x7fd01d6461b6 in Sketcher::SketchObject::buildShape() from /tmp/.mount_freecaqu4dhg/usr/lib/Sketcher.so+0x236
#4  0x7fd01d6471ea in Sketcher::SketchObject::execute() from /tmp/.mount_freecaqu4dhg/usr/lib/Sketcher.so+0xda
#5  0x7fd0dbd3e552 in App::DocumentObject::recompute() from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADApp.so+0xc2
#6  0x7fd06105a3cc in Part::Feature::recompute() from /tmp/.mount_freecaqu4dhg/usr/lib/./Part.so+0xc
#7  0x7fd0dbd1873c in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADApp.so+0x32c
#8  0x7fd0dbd24ae5 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADApp.so+0x625
#9  0x7fd0dbda31ae in App::DocumentPy::recompute(_object*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADApp.so+0x2ee
#10  0x7fd0dbda32c4 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADApp.so+0x24
#11  /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7fd0dc30a068]
#12  /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7fd0dc2e9b5c]
#13  /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7fd0dc2f5532]
#14  /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7fd0dc3afd1d]
#15  /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7fd0dc3af794]
#16  /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7fd0dc3cebf3]
#17  /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7fd0dc3ca69a]
#18  /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7fd0dc3be9c9]
#19  0x7fd0db98b48c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADBase.so+0x6c
#20  0x7fd0dcc54964 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x94
#21  0x7fd0dcc54af3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0xe3
#22  0x7fd0dcc54c14 in Gui::Command::updateActive() from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x44
#23  0x7fd01c5622d3 in SketcherGui::tryAutoRecompute(Sketcher::SketchObject*, bool&) from /tmp/.mount_freecaqu4dhg/usr/lib/SketcherGui.so+0x73
#24  0x7fd01c5623c3 in SketcherGui::tryAutoRecomputeIfNotSolve(Sketcher::SketchObject*) from /tmp/.mount_freecaqu4dhg/usr/lib/SketcherGui.so+0x23
#25  /tmp/.mount_freecaqu4dhg/usr/lib/SketcherGui.so(+0x21282a) [0x7fd01c5f182a]
#26  /tmp/.mount_freecaqu4dhg/usr/lib/SketcherGui.so(+0x212e84) [0x7fd01c5f1e84]
#27  /tmp/.mount_freecaqu4dhg/usr/lib/SketcherGui.so(+0x1a6540) [0x7fd01c585540]
#28  0x7fd01c74086d in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /tmp/.mount_freecaqu4dhg/usr/lib/SketcherGui.so+0x1fd
#29  0x7fd0dd02f7e0 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x160
#30  0x7fd0d9cf004e in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0xbe
#31  0x7fd0d9aeee38 in SoAction::traverse(SoNode*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0x68
#32  0x7fd0d9ca8cca in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0x15a
#33  0x7fd0d9cf8ce8 in SoGroup::doAction(SoAction*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0x58
#34  0x7fd0d9d13f8a in SoSeparator::doAction(SoAction*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0x2a
#35  0x7fd0d9aeee38 in SoAction::traverse(SoNode*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0x68
#36  0x7fd0d9ca8cca in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0x15a
#37  0x7fd0d9cf8ce8 in SoGroup::doAction(SoAction*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0x58
#38  0x7fd0d9d13f8a in SoSeparator::doAction(SoAction*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0x2a
#39  0x7fd0d9aeee38 in SoAction::traverse(SoNode*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0x68
#40  0x7fd0d9af8a2c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0xac
#41  0x7fd0d9aefb1a in SoAction::apply(SoNode*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0x2ea
#42  0x7fd0d9cc4dd2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/./libCoin.so.80+0x72
#43  0x7fd0dcfb4a7e in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x1e
#44  0x7fd0dcfbfee8 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x58
#45  0x7fd0dcfcc170 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x40
#46  0x7fd0dcfd299f in Gui::TouchpadNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x2df
#47  0x7fd0dcfcc405 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x65
#48  0x7fd0dcff4a6b in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x6b
#49  0x7fd0dcfab2a5 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x1e5
#50  0x7fd0da80a8bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libQt5Core.so.5+0x7b
#51  0x7fd0db33739e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#52  0x7fd0db33e8f9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libQt5Widgets.so.5+0xa39
#53  0x7fd0dcc09c69 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x99
#54  0x7fd0da80ab28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libQt5Core.so.5+0x108
#55  0x7fd0db33d3eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#56  /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7fd0db38de6f]
#57  /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7fd0db3910c9]
#58  0x7fd0db3373ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#59  0x7fd0dcc09c69 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x99
[... cut ...]
#68  0x7fd0da80970b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libQt5Core.so.5+0x11b
#69  0x7fd0da811212 in QCoreApplication::exec() from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libQt5Core.so.5+0x82
#70  0x7fd0dcb6c3a2 in Gui::Application::runApplication() from /tmp/.mount_freecaqu4dhg/usr/bin/../lib/libFreeCADGui.so+0x972
#71  /tmp/.mount_freecaqu4dhg/usr/bin/freecad(+0x7620) [0x563a97637620]
#72  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fd0da136083]
#73  /tmp/.mount_freecaqu4dhg/usr/bin/freecad(+0x7981) [0x563a97637981]

kevinsmia1939, 2024-07-07:
Can confirm this crash.
Create new body in part design > create new sketch > add a point
Crash immediately.
```
kevin@ptr-1v5mxqpzcs27pz5pbar:~/Downloads> ./FreeCAD_weekly-builds-38043-conda-Linux-x86_64-py311.AppImage
FreeCAD 0.22.0, Libs: 0.22.0devR38043 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

_IceTransSocketUNIXConnect: Cannot connect to non-local host localhost.localdomain
_IceTransSocketUNIXConnect: Cannot connect to non-local host localhost.localdomain
Qt: Session management error: Could not open network socket
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x41240) [0x7efebd241240]
#1  0x7efebf67eead in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADApp.so+0xed
#2  0x7efebf67fc97 in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADApp.so+0x297
#3  0x7efe95ef91b6 in Sketcher::SketchObject::buildShape() from /tmp/.mount_FreeCARbVu06/usr/lib/Sketcher.so+0x236
#4  0x7efe95efa1ea in Sketcher::SketchObject::execute() from /tmp/.mount_FreeCARbVu06/usr/lib/Sketcher.so+0xda
#5  0x7efebf45d552 in App::DocumentObject::recompute() from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADApp.so+0xc2
#6  0x7efea2ea23cc in Part::Feature::recompute() from /tmp/.mount_FreeCARbVu06/usr/lib/./Part.so+0xc
#7  0x7efebf43773c in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADApp.so+0x32c
#8  0x7efebf443ae5 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADApp.so+0x625
#9  0x7efebf4c21ae in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADApp.so+0x2ee
#10  0x7efebf4c22c4 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADApp.so+0x24
#11  /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7efebfc06068]
#12  /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7efebfbe5b5c]
#13  /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7efebfbf1532]
#14  /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7efebfcabd1d]
#15  /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7efebfcab794]
#16  /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7efebfccabf3]
#17  /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7efebfcc669a]
#18  /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7efebfcba9c9]
#19  0x7efebef3d48c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADBase.so+0x6c
#20  0x7efec0587964 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADGui.so+0x94
#21  0x7efec0587af3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADGui.so+0xe3
#22  0x7efec0587c14 in Gui::Command::updateActive() from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADGui.so+0x44
#23  0x7efe245832d3 in SketcherGui::tryAutoRecompute(Sketcher::SketchObject*, bool&) from /tmp/.mount_FreeCARbVu06/usr/lib/SketcherGui.so+0x73
#24  0x7efe245833c3 in SketcherGui::tryAutoRecomputeIfNotSolve(Sketcher::SketchObject*) from /tmp/.mount_FreeCARbVu06/usr/lib/SketcherGui.so+0x23
#25  /tmp/.mount_FreeCARbVu06/usr/lib/SketcherGui.so(+0x21282a) [0x7efe2461282a]
#26  /tmp/.mount_FreeCARbVu06/usr/lib/SketcherGui.so(+0x212e84) [0x7efe24612e84]
#27  /tmp/.mount_FreeCARbVu06/usr/lib/SketcherGui.so(+0x1a6540) [0x7efe245a6540]
#28  0x7efe2476186d in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /tmp/.mount_FreeCARbVu06/usr/lib/SketcherGui.so+0x1fd
#29  0x7efec09627e0 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADGui.so+0x160
#30  0x7efebccc704e in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/./libCoin.so.80+0xbe
#31  0x7efebcac5e38 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/./libCoin.so.80+0x68
#32  0x7efebcc7fcca in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/./libCoin.so.80+0x15a
#33  0x7efebcccfce8 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/./libCoin.so.80+0x58
[... cut ...]
#69  0x7efebd8bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libQt5Core.so.5+0x82
#70  0x7efec049f3a2 in Gui::Application::runApplication() from /tmp/.mount_FreeCARbVu06/usr/bin/../lib/libFreeCADGui.so+0x972
#71  /tmp/.mount_FreeCARbVu06/usr/bin/freecad(+0x7620) [0x560ee1e8c620]
#72  /lib64/libc.so.6(+0x2a1f0) [0x7efebd22a1f0]
#73  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7efebd22a2b9]
#74  /tmp/.mount_FreeCARbVu06/usr/bin/freecad(+0x7981) [0x560ee1e8c981]

```

TheVirak, 2024-07-10:
Similar issue in FreeCAD_weekly-builds-38079-conda-Linux-x86_64-py311.AppImage, however the crash only happens after closing the sketch.

## Steps to reproduce

- Create new Parametric Part
- Create body
- Create sketch, does not matter on which plane
- Create point anywhere, does not matter if it is constrained or not
- Close sketch

## Expected outcome

Sketch closes, return to PartDesign workspace

## Observed outcome

FreeCAD crashes:

```
FreeCAD 0.22.0, Libs: 0.22.0devR38079 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x40d00) [0x7f89ef64fd00]
#1  0x7f89f187eead in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADApp.so+0xed
#2  0x7f89f187fc97 in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADApp.so+0x297
#3  0x7f89b02f91b6 in Sketcher::SketchObject::buildShape() from /tmp/.mount_FreeCA4qrpFE/usr/lib/Sketcher.so+0x236
#4  0x7f89b02fa1ea in Sketcher::SketchObject::execute() from /tmp/.mount_FreeCA4qrpFE/usr/lib/Sketcher.so+0xda
#5  0x7f89f165d552 in App::DocumentObject::recompute() from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADApp.so+0xc2
#6  0x7f89bd4a23cc in Part::Feature::recompute() from /tmp/.mount_FreeCA4qrpFE/usr/lib/./Part.so+0xc
#7  0x7f89f163773c in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADApp.so+0x32c
#8  0x7f89f1643ae5 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADApp.so+0x625
#9  0x7f89f16c21ae in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADApp.so+0x2ee
#10  0x7f89f16c22c4 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADApp.so+0x24
#11  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7f89f1e06068]
#12  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f89f1de5b5c]
#13  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f89f1df1532]
#14  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7f89f1eabd1d]
#15  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f89f1eab794]
#16  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7f89f1ecabf3]
#17  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7f89f1ec669a]
#18  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7f89f1eba9c9]
#19  0x7f89f113d48c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADBase.so+0x6c
#20  0x7f89f2787924 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADGui.so+0x94
#21  0x7f89f2787ab3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADGui.so+0xe3
#22  0x7f89f2787bd4 in Gui::Command::updateActive() from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADGui.so+0x44
#23  0x7f895e157ac7 in SketcherGui::ViewProviderSketch::unsetEdit(int) from /tmp/.mount_FreeCA4qrpFE/usr/lib/SketcherGui.so+0x317
#24  0x7f89f2b59b73 in Gui::ViewProvider::finishEditing() from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADGui.so+0x13
#25  0x7f89f26fcc6d in Gui::Document::_resetEdit() from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADGui.so+0x9d
#26  0x7f89f267c56a in Gui::Application::setEditDocument(Gui::Document*) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADGui.so+0x3a
#27  0x7f89f2711710 in Gui::DocumentPy::resetEdit(_object*) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADGui.so+0x30
#28  0x7f89f2711744 in Gui::DocumentPy::staticCallback_resetEdit(_object*, _object*) from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADGui.so+0x24
#29  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7f89f1e06068]
#30  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f89f1de5b5c]
#31  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f89f1df1532]
#32  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7f89f1eabd1d]
#33  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f89f1eab794]
#34  /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7f89f1ecabf3]
[... cut ...]
#68  0x7f89efabd212 in QCoreApplication::exec() from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libQt5Core.so.5+0x82
#69  0x7f89f269f362 in Gui::Application::runApplication() from /tmp/.mount_FreeCA4qrpFE/usr/bin/../lib/libFreeCADGui.so+0x972
#70  /tmp/.mount_FreeCA4qrpFE/usr/bin/freecad(+0x7620) [0x56318b4da620]
#71  /lib64/libc.so.6(+0x2a088) [0x7f89ef639088]
#72  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7f89ef63914b]
#73  /tmp/.mount_FreeCA4qrpFE/usr/bin/freecad(+0x7981) [0x56318b4da981]
```

adrianinsaval, 2024-07-13:
I can reproduce this on freecad-daily from the ppa and on my archlinux build, which means this is not a bundle issue. This is the wrong repo to report this

adrianinsaval, 2024-07-13:
I think this was reported here: https://github.com/FreeCAD/FreeCAD/issues/15154

please open a new issue in that repo if you think this is a different issue. This repo is specifically about packaging. Nobody is watching for bugs in the software here

closed 2024-07-13 by adrianinsaval