FreeCAD/FreeCAD-Bundle #263: Sketcher External Geometry without ExternalGeometryExtension

erwanyves, 2024-07-07:
while trying to get external geometry from another sketch, getting this error message: " Sketcher External Geometry without ExternalGeometryExtension: 09e268ac-8f81-475b-baf7-8cf825dbbcdf ". in some case, impossible to pad it.

RWL-69, 2024-07-08:
Confirmed with FreeCAD_weekly-builds-38043-conda-Linux-x86_64-py311.AppImage

> 
```
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Body
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Sketch
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Sketch001
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad001
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad001
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Sketch002
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad002
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad002
<PropertyLinks> PropertyLinks.cpp(433): Fan_curtain_holder#Sketch001.AttachmentSupport auto change element reference Fan_curtain_holder#Pad ;g73;SKT;:H7d9,E;FAC;:H7d9:4,F;:G0;XTR;:H7d9:8,F.Face14 -> ;g7;SKT;:H7d9,E;FAC;:H7d9:4,F;:G0;XTR;:H7d9:8,F.Face14
<PropertyLinks> PropertyLinks.cpp(433): Fan_curtain_holder#Sketch001.ExternalGeometry auto change element reference Fan_curtain_holder#Pad ;g898;SKT;:H7d9,E;:G;XTR;:H7d9:7,F;:U;XTR;:H7d9:7,E.Edge16 -> ;g15;SKT;:H7d9,E;:G;XTR;:H7d9:7,F;:U;XTR;:H7d9:7,E.Edge16
Sketcher External Geometry without ExternalGeometryExtension: 46ca8bb2-5e01-47e1-aca9-4e444f4dae37 
Sketcher External Geometry without ExternalGeometryExtension: c7f5bb94-ad94-43b3-8a9e-721c2e64b98f 
Sketcher External Geometry without ExternalGeometryExtension: 3ceaeb6f-666b-4d9a-9d7b-4505bf33a9d3 
Sketcher External Geometry without ExternalGeometryExtension: 2205d1c5-1f93-4e9b-a2fb-ea9cd9de1c48
```

In FreeCAD_weekly-builds-38001-conda-Linux-x86_64-py311.AppImage, trying the same thing, I get 
```
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Body
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Sketch
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Pad
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Pad
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Sketch001
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Pocket
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Pocket
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Sketch002
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Pad001
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Pad001
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Sketch003
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Pocket001
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Curtain_pull#Pocket001
Reading failed from embedded file: StringHasher.Table.txt (1823 bytes, 260 bytes compressed)
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Body
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Sketch
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Sketch001
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad001
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad001
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Sketch002
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad002
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: Fan_curtain_holder#Pad002
<PropertyLinks> PropertyLinks.cpp(433): Fan_curtain_holder#Sketch001.AttachmentSupport auto change element reference Fan_curtain_holder#Pad ;g73;SKT;:H7d9,E;FAC;:H7d9:4,F;:G0;XTR;:H7d9:8,F.Face14 -> ;g46;SKT;:H7d9,E;FAC;:H7d9:4,F;:G0;XTR;:H7d9:8,F.Face14
<PropertyLinks> PropertyLinks.cpp(433): Fan_curtain_holder#Sketch001.ExternalGeometry auto change element reference Fan_curtain_holder#Pad ;g898;SKT;:H7d9,E;:G;XTR;:H7d9:7,F;:U;XTR;:H7d9:7,E.Edge16 -> ;g54;SKT;:H7d9,E;:G;XTR;:H7d9:7,F;:U;XTR;:H7d9:7,E.Edge16
6.9e-08 <App> Document.cpp(3152): exception in Fan_curtain_holder#Sketch001 thrown: argument not found
Sketch001: argument not found
57.7275 <App> Document.cpp(3152): exception in Fan_curtain_holder#Sketch001 thrown: argument not found
Sketch001: argument not found
57.8322 <App> Document.cpp(3152): exception in Fan_curtain_holder#Sketch001 thrown: argument not found
Sketch001: argument not found
```
Didn't have any time to investigate.

adrianinsaval, 2024-07-14:
https://github.com/FreeCAD/FreeCAD/issues/15117

closed 2024-07-14 by adrianinsaval