FreeCAD/FreeCAD-Bundle #264: File icon is not being displayed on AppImage anymore

leoheck, 2024-07-08:
This issue was first posted here https://github.com/FreeCAD/FreeCAD/issues/15222

File associations for FCStd are not being recognized anymore. 
Also, .FCSbak can also have a specific association.

```
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

adrianinsaval, 2024-07-08:
> Also, `.FCSbak` could also have a specific association and a custom icon.

not somethihg to be discussed here and IIRC freecad can't even open fcbak files on puprose

adrianinsaval, 2024-07-08:
I have transferred your original issue to this repo since it has more information than this one. Therefore I will close this

closed 2024-07-08 by adrianinsaval

leoheck, 2024-07-08:
> not somethihg to be discussed here and IIRC freecad can't even open fcbak files on puprose

does not matter,  if the extension is a freecad thing, it could have its own icon. I am not talking about who is going open it, or who can open it.