FreeCAD/FreeCAD-Bundle #265: Nautilus does not know the .FCStd extension anymore.

leoheck, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Nautilus does not know the FCStd extension anymore.

[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.12
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * SheetMetal 0.4.20
  * fasteners 0.5.24
  * asm4_disable_version_check_popup
  * Render 2024.7.4
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-27:
We need a GNOME person to weigh-in

leoheck, 2024-08-27:
Freecad icons look like this for me.
[image]

For instance, these are Kicad icons and they look great.
[image]

I could help to fix this, but I am kinda running against clock to finish a project. And Freecad 0.22 is also giving all sorts of issues. If I can advance with my side project, I am going to explore how to fix this.

luzpaz, 2024-12-11:
Is this still a thing ?

furgo16, 2024-12-11:
I still see the generic (non-FreeCAD) icon as in https://github.com/FreeCAD/FreeCAD-Bundle/issues/265#issuecomment-2313172161. I'm using Ubuntu 22.04 on my laptop and the latest .snap and .deb daily packages.

probonopd, 2024-12-16:
Probably you would have to copy a `.desktop` file into `$HOME` that would associate the MIME type with the AppImage. But the whole `.desktop` file idea is not really well suitable for portable applications ("bundles") that can come and go (and be there in multiple versions). Imho the whole Linux XDG specifications need to be rethrought from scratch with bundles in mind, more like the Mac does it.

adrianinsaval, 2024-12-20:
I'm not in favor of automatic creation of desktop files by the appimage, presumably the reporter was using appimagelauncher though so a desktopfile should have been created, I guess the problem is that appimagelauncher does not copy over the document icon, only the app icon.

leoheck, 2024-12-20:
Exactly, and also another problem is that since some weeks ago the appimagelauncher is not working anymore which is really sad. 

The Gear Lever sucks and also does not allow me to have 2 "installed" appimages of the same tool (for 1.0 and nightlies for instance). Or maybe this is still something that can be tweaked in the AppImage. But right now, installing one overrides the other. And also, the desktop file is named after the git commit which is horrible for the user to identify the version to decide which version they want to launch...

luzpaz, 2025-01-04:
2 responses from the forum:

> File type associations are not maintained by GNOME: GNOME consumes the data in the [shared-mime-info database](https://gitlab.freedesktop.org/xdg/shared-mime-info).
>
> It’s entirely possible that FreeCAD is shipping its own extensions to that database, which is allowed; but AppImage is a self-contained distribution format that does not interact with the system in any way (which is why it does not provide desktop files either, and requires manual installation of files by the user), so it’s unlikely that it’ll work for MIME type extensions, or icons.

> My strong recommendation is to stop shipping AppImages, unless you’re also willing to document how to integrate them into a freedesktop system, something that the AppImage development community is wholly uninterested in doing, seeing as they are trying to replicate the macOS app deployment mechanism.

And 

> What is result of the following command

`gio info -a "*" myfile.FCStd | grep icon`

luzpaz, 2025-01-05:
`gio info -a "*" myfile.FCStd | grep icon`  
Need a GNOME user to test this out.

leoheck, 2025-01-05:
Hi @luzpaz this is the output

```
➜ gio info -a "*" myfile.FCStd | grep icon

  standard::icon: application-x-extension-fcstd, application-x-extension-fcstd-symbolic
  standard::symbolic-icon: application-x-extension-fcstd-symbolic, application-x-extension-fcstd-symbolic, application-x-extension-fcstd
```

AbuShawarib, 2025-01-05:
Can you see the icon if you change to `hicolor` icon theme?

AbuShawarib, 2025-01-05:
Pretty much what said by nazar-pc and the quoted forum post. Linux XDG standards expects apps to install MIME types (if they are not registered in the general shared MIME DB) and that icons be installed in the usual theme paths. Both don't happen magically with appimages.

adrianinsaval, 2025-01-05:
I don't think this ever worked, if it did it was probably only by chance, I don't think appimagelauncher copies the doc icon, only the application icon.

luzpaz, 2025-01-05:
This did work because @leoheck custom made a logo for the FreeCAD Conda AppImages, IIRC

adrianinsaval, 2025-01-05:
I believe that was the app icon, not document icon. The app icon might have been used for documents but that's incorrect