FreeCAD/FreeCAD-Bundle #269: Windows and Display gone wild (not displaying correctly/partially, placed odd, not painting)

longrackslabs, 2024-07-13:
Not sure what happened, but somehow FreeCad 0.22 started having probems displaying/rending windows.  The windows suddendly moved, aren't painting / udpating,etc.  

This renders the app unusable.

Using freecad weekly version
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git) AppImage
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.6
  * Curves 0.6.22
  * fasteners 0.5.0
  * ExplodedAssembly
  * A2plus 0.4.62
  * lattice2 1.0.0
[image]

adrianinsaval, 2024-07-13:
try cleaning your config and/or setting a theme in preferences. This looks like a problem coming from your system theme

longrackslabs, 2024-07-13:
I literally just installed this.  I don’t know about themes.  I am using
the tool with all defaults

adrianinsaval, 2024-07-14:
that does not look default, it seems you have enabled transparent view on your panels which does not work so well when using system themes. Please try removing/renaming `~/.config/FreeCAD` or setting a theme in the application

longrackslabs, 2024-07-14:
Like i said, didn't change any settings or preferences - I was just using the app normally., so not sure how I would have 'enabled' anything.  

Anayway, I finally deleted .config/FreeCad director and reinstalled and things are fixed now.  If this happens again i'll save my config dir and attach.

adrianinsaval, 2024-07-14:
It's a small button on the panel. you might have miss clicked, you can probably use better the feature if you enable a theme in freecad

longrackslabs, 2024-07-14:
which button on which panel?

maxwxyz, 2024-07-25:
Is it still present in the latest weekly?

longrackslabs, 2024-07-28:
I installed this weeks build and had the same windowing problems.  I did uninstall older versions (I am on mac).  I was able to fix the window rendering by Reset All in preferences.  This also means all my preferences were lost, but it did make the tool usable.

closed 2024-08-12 by maxwxyz