FreeCAD/FreeCAD-Bundle #270: Crashes on Mesa >0.24

RandomLegend, 2024-07-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Appimages for FreeCAD crash on Linux with AMD cards running Mesa >0.24 and require Mesa downgrade to 23.5 to run.

Error produced is:

```
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: undefined symbol: amdgpu_va_get_start_addr (search paths /usr/lib/dri, suffix _dri)
failed to load driver: radeonsi
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: undefined symbol: amdgpu_va_get_start_addr (search paths /usr/lib/dri, suffix _dri)
failed to load driver: swrast
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Custom folder not found: /mnt/Wyrm/Nextcloud/Kreativ/Drucker, Laser, CNC/DruckerFailed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
/tmp/.mount_FreeCAmhbENP/AppRun: line 43: 77083 Aborted                 (core dumped) ${MAIN} "$@"
```

And this happens on all forks and versions. Whether it be the stable version, dev weekly build or the Ondsel fork.

### Full version info

```shell
Can't provide that info as the application doesn't start.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-07-14:
> The problem was also reported in realthunder's branch and a solution/workaround posted, [that I can confirm works for me](https://github.com/realthunder/FreeCAD/issues/960#issuecomment-1974233355).
> 
> Apparently there are `libdrm*` files included in the AppImage, that interfere with system libraries. Removing those and repacking seems to solve the problem.

should these be removed from the appimage that we ship or are these libraries that will be needed in some systems but be problematic on others?

rodlie, 2024-07-29:
Simple rule for deploying on Linux: Never replace/bundle system libraries.

You by default break Mesa by shipping certain libraries, that it works on some distros are just luck in regards to libstdc compat.

Drop everything regarding GL and gcc.