FreeCAD/FreeCAD-Bundle #271: "No autoconstraints added, please report" message received.

shadowbane1000, 2024-07-15:
[panels.zip](https://github.com/user-attachments/files/16229570/panels.zip)

### Bug:
When creating a slot with autoconstraints on, in the particular situation of the sketch above, the following message is received:
20:07:29  Autoconstraints cause redundancy. Removing them
20:07:29  Redundant constraint is not an autoconstraint. No autoconstraints or additional constraints were added. Please report!
And the slot does not appear in the view.

### Expected result:
The slot should have been created with autoconstraints.

### Steps to reproduce:

1. Using FreeCAD_weekly-builds-38127-conda-Windows-x86_64-py311
2. Open the attached project file.
3. Edit Sketch001
4. In the bottom right corner of the panel find a reference slot feature [image]
5. delete the two arcs and two lines that make up the feature.
6. Make sure auto constraints is turned on.
7. recreate the slot by clicking first on the center of the left circle, then the center of the right circle, then somewhere along the edge of about where the slot belongs.
8. The error should now be displayed.

JohnOCFII, 2024-07-15:
I've seen with 38127 on Mac as well.

maxwxyz, 2024-07-18:
I'll transfer the issue to the FreeCAD repo, please add full about info.