FreeCAD/FreeCAD-Bundle #273: Install V # 0.21.2 on Win 7, 64, Sys fails

BreiteGu, 2024-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Install V # 0.21.2 on Win 7, 64, Sys fails
Installing on other than "C:" drive (keeping more than a single live version) fails. The install will abort trying to operate on the C: drive whereas destination was set to D:

### Full version info

```shell
Intaling V 0.21.2 parallel to 19.n for testing / evaluation ....
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

I agree

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-28:
I cannot reproduce.
@adrianinsaval FYI

BreiteGu, 2024-07-28:
Very simple. It happens during install, every time I try.
My computer sports several HD (e.g. C, D. E aso)
D: contains (as does C:) the folders: D:\Program Files and D:\Program Files (x86)
During install, the Installer proposes to install to Directory C:\....
now if you change this to D:\.... and leave the remainder untouched, the error happens, Installer aborts with error

adrianinsaval, 2024-08-05:
if you will make an install for all users you must give administrator privileges, if you don't want that please choose the option to install for the current user only. The error you show seems to be about the installer trying to setup the thumbnailer for fcstd file preview on the file explorer. I will also say that windows 7 is not really supported anymore so use at your own risk. Also that particular file from the error message is not needed to run freecad so feel free to ignore it and the installed software should probably still work

closed 2024-10-05 by luzpaz