FreeCAD/FreeCAD-Bundle #274: A2PLUS

Luciniki, 2024-07-28:
Buongiorno,

stò riscontrando una grande lentezza nel risolvere il posizionamento dei miei componenti.
Questo problema lo riscontro nei rilasci settimanali successivi a quello qui sotto:

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.65
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.23

Ripeto per chiarezza: Il rilascio 38261 funziona bene in modo RAPIDO, mentre quelli successivi sono molto LENTI.

Non so esattamente su quale versione è comparso il problema.

Grazie del lavoro che fate, questo progetto è bellissimo!

luzpaz, 2024-07-28:
English please

Luciniki, 2024-07-30:
With version 38375 it is still very slow.
for the placement of a component I timed 27.2 seconds to complete the operation.

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * A2plus 0.4.65

The same identical operation done with version 38261 took 3.5 seconds

Luciniki, 2024-08-07:
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * A2plus 0.4.65
  
**with this version it went fine and came back fast**

Grazie!

Luciniki, 2024-08-07:
other small errors but I know you are working on it

closed 2024-08-07 by Luciniki

luzpaz, 2024-08-18:
Please report A2plus issues to https://github.com/kbwbe/A2plus