FreeCAD/FreeCAD-Bundle #277: Mesh Workbench: Quasi structured quad not working

jintat87, 2024-08-04:
Hi,

I'm trying on the FreeCAD_weekly-builds-0.22.0dev-38385.

In Mesh Workbench, when I use Tesselate shape to create mesh from shape with Gmsh using Quasi Structured Quad, it produces no result.

I have tried the gmsh that comes with the package in bin folder, as well as latest gmsh-4.13.1. Both same without result.

Hope you can look into this. Thanks.

luzpaz, 2024-08-04:
Full About info please. Also is this Appimage (packaging) related or just FreeCAD itself ?

jintat87, 2024-08-05:
it's regarding this latest fix weekly build [#15661](https://github.com/FreeCAD/FreeCAD/pull/15661
)
The Quasi-structured Quad added to Mesh wb but there's no output.
[image]

luzpaz, 2024-08-07:
@jintat87 is this an appimage related bug or is it a bug in FreeCAD ?

maxwxyz, 2024-08-12:
If this issue is still existing, please open it in the FreeCAD repo if there are no similar issues/duplicates.
@FEA-eng FYI

closed 2024-08-12 by maxwxyz