FreeCAD/FreeCAD-Bundle #279: Snap build failed: Exported credentials are no longer valid for the Snap Store

furgo16, 2024-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Publishing the latest daily snap package build failed with:

```
Exported credentials are no longer valid for the Snap Store.
Recommended resolution: Run export-login and update SNAPCRAFT_STORE_CREDENTIALS.
For more information, check out: https://snapcraft.io/docs/snapcraft-authentication
Full execution log: '/home/runner/.local/state/snapcraft/log/snapcraft-20240[8](https://github.com/FreeCAD/FreeCAD-snap/actions/runs/10240590888/job/28327659024#step:5:9)05-020312.844130.log'
```

Log: https://github.com/FreeCAD/FreeCAD-snap/actions/runs/10240590888/job/28327659024

### Full version info

```shell
N/A
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-05:
@adrianinsaval FYI

chennes, 2024-08-05:
Probably really @luzpaz, @yorikvanhavre, or @chennnes 😄 -- I think we're the ones with access to the snap store.

adrianinsaval, 2024-08-05:
also this is not the repo for this either. I can't seem to transfer it to FreeCAD-Snap though