FreeCAD/FreeCAD-Bundle #281: Update installers to register new FreeCAD logo for FCStd / FCMacro in different OS

maxwxyz, 2024-08-06:
@adrianinsaval you've mentioned it yesterday, I hope I've captured it correctly:
Update installers to register new FreeCAD logo for FCStd / FCMacro in different OS

adrianinsaval, 2024-08-06:
I'm not entirely sure what's the mechanism by which the logo gets added to the preview on windows, I don't know if it's a feature from windows itself or if it's from our thumbnail provider.

I don't even know how to compile the dll for the thumbnails actually. @chennes do you know how to do that?

maxwxyz, 2024-10-12:
Can this be closed or was there any feedback from the RCs that this is not working?

adrianinsaval, 2024-10-12:
From my testing the overlayed icon seems to be added by windows itself based on what program is configured to open the file so there's no action needed

closed 2024-10-13 by maxwxyz