FreeCAD/FreeCAD-Bundle #283: macOS: Released disk image for arm mac requires rosetta

jmgurney, 2024-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The dmg that claims to support M1/M2 arm Macs does not work w/o Rosetta.  It pops open a dialog telling me to install rosetta.

### Full version info

```shell
I cannot use about box since FreeCAD refuses to run, but I have tried `FreeCAD-0.21.0-mac-arm64.dmg`, `FreeCAD-0.21.1-mac-arm64.dmg` and `FreeCAD-0.21-RC1-mac-arm64.dmg`.
```

### Subproject(s) affected?

None

### Anything else?

Machine is a 2021 14-inch MBP w/ an M1 Pro running macOS 14.0 (23A344).  It does not have rosetta installed, partly to discover programs which claim to support ARM, but actually do not.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2024-08-13:
> Could you please test the latest weekly?

That seems to have resolved this issue.  I normally have Rosetta run as needed, but by looking at Activity Monitor when launching last weeks' build, I could see that `bash` ran as Intel.  Now with tonight's build (38459) `bash` is shown as associated with FreeCAD and running as an Apple binary. 

[image]

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.8.10
```

adrianinsaval, 2024-08-13:
Great! I'll close the issue then

closed 2024-08-13 by adrianinsaval

dansharkey, 2024-08-14:
I've just tested the latest weekly (2d5a84e014483d98906486b8266196d000632d614c9e9218129a4cc4a24f0932  FreeCAD_weekly-builds-38459-conda-macOS-arm64-py311.dmg) and it is still requesting Rosetta even with LSArchitecturePriority=arm64 in the Plist file.

I think the best way to check with Rosetta installed is to look at the "Get Info" output after right-clicking on the .app file. Under "Kind" it should show as "Application (Apple silicon)" (as per this screenshot contrasting FreeCAD with Blender):
[image]
This is my system monitor after running the bash script directly from the command line
(`cd /Applications/FreeCAD.app/Contents; ./MacOS/FreeCAD`)
[image]

Voting to re-open this issue (sorry!)

feheragoston, 2024-08-14:
It works for me, I'm on Ventura 13.6.4. I don't have Rosetta installed and for the first time I was able to run FreeCAD without using the terminal.

@dansharkey is correct that on the info screen the Kind is `Application` where for other apps it is either `Application (Universal)` or `Application (Apple silicon)`.

oursland, 2024-08-15:
A possible solution would be to create a universal binary by installing both the `x86-64` and `arm64` conda packages as `FreeCAD.app/Resources/${arch}/`, then create a simple C++ launcher to replace the shell script that configures the environment and launches FreeCAD.  CMake can and build the launcher for both platforms and generate the output as a Universal Binary.  I believe this would address the issues raised on https://github.com/FreeCAD/FreeCAD-Bundle/issues/283.

References:

* [How to configure python conda Environments for both arm64 and x86_64 on M1 Apple Silicon](https://stackoverflow.com/a/73839226)
* [Example macOS universal app with cmake](https://github.com/faaxm/exmpl-cmake-universal-app)
    * [CMake: `CMAKE_OSX_ARCHITECTURES`](https://cmake.org/cmake/help/latest/variable/CMAKE_OSX_ARCHITECTURES.html)

saraelsa, 2024-08-16:
I am also unable to launch the latest weekly build (38459) without Rosetta. I'm on Sonoma 14.6.1.