FreeCAD/FreeCAD-Bundle #285: ❗Please do not open general FreeCAD issues (read first)

maxwxyz, 2024-08-14:
This repository is only for bundle-related issues.
Please do **not** open general issues for FreeCAD here.
### Instead use the FreeCAD repository https://github.com/FreeCAD/FreeCAD/issues

1. Go to the [FreeCAD](https://github.com/FreeCAD/FreeCAD/issues) repository
2. Search for similar / already existing issues or potential duplicates
3. Use the most updated stable or [development versions](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) of FreeCAD
4. Post version info from e.g. Help > About FreeCAD > Copy to clipboard;
5. Start recording a macro Macro > Macro recording... and repeat all steps. Stop recording after the issue occurs and upload the saved macro or copy the macro code in the issue
6. Post a Step-By-Step explanation on how to recreate the issue
7. Upload an example file (FCStd as ZIP file) to demonstrate the problem

FeedTheChunk, 2024-08-14:
Max can you expand on what "bundle-relate issues" are?  Is this errors encountered with the Dev version?

maxwxyz, 2024-08-14:
Done