FreeCAD/FreeCAD-Bundle #286: Build 38459 for macOS x86 will not open

sslupsky, 2024-08-14:
There might be some sort of signing issue with the 38459 build for x86 on macOS.  When I attempt to open it macOS throws an error which looks suspiciously like a gatekeeper issue.

[image]

I am running macOS Sonoma 14.6.1

luzpaz, 2024-08-15:
As a workaround, You need to right click the freecad Icon (or press shift and press the freecad icon) to invoke the application to bypass the annoying macOS gatekeeping issue.

@JohnOCFII is this still an issue on macOS for you? Was @chennes able to successfully sign macOS builds ?

JohnOCFII, 2024-08-15:
> @JohnOCFII is this still an issue on macOS for you? Was @chennes able to successfully sign macOS builds ?

I've only tried this version on M1/arm Macs, and the most recent version worked fine.  The team **did** successfully sign a build, but (for some reason) decided NOT to sign the weeklies, and plan to only sign released versions.

There _was_ work in this area for this release to try to have one of the launching processes use an -arm- based version of `bash` instead of an x86 version of bash.  That tested properly on an arm Mac, but potentially that now created an issue for x86 Macs.  Not sure.  See this other Github issue: https://github.com/FreeCAD/FreeCAD-Bundle/issues/283#issuecomment-2289771832

oursland, 2024-08-23:
The macOS release for Intel processors is operational again.  You will need to use the right mouse click and select "Open" to permit running the unsigned file.

The macOS release for Apple Silicon is still misbehaving.

sslupsky, 2024-08-23:
Thank you.  I downloaded build 38495 and I can open in on an Intel Mac.

closed 2024-08-23 by sslupsky