FreeCAD/FreeCAD-Bundle #288: Can we trigger a new build past 38459?

leoheck, 2024-08-16:
Is it possible to update the weekly right now so the version will move from `38459`? I could not build it myself and this version has some bad issues (they are not usable) with the Part Design Mirror/Transform/Multitransform features? I apprecisate if someone could do that.

luzpaz, 2024-08-17:
CC @adrianinsaval

leoheck, 2024-08-17:
Guys, I could build it last night, I just did not tested it yet  I was trying to use qt6 but it is still being ported. with qy5 everything went well. I just had to install one or two packages that the wiki doesn't knowledge. I will check if I can update the page somehow. If this upgrade was not built here yet it is still worth so it won't stop other users of using it.

closed 2024-08-18 by leoheck