FreeCAD/FreeCAD-Bundle #291: weekly build 38467 for MacOS is damaged

smr02, 2024-08-19:
FreeCAD_weekly-builds-38467-conda-macOS-arm64-py311.dmg

[image]

JohnOCFII, 2024-08-19:
Same experience: 
[image]

MisterMakerNL, 2024-08-19:
Is this how Apple solves all issues? XD

BillTuttle, 2024-08-19:
Is this the arm or intel package? Intel doesn't seem to be damaged, but does now require MacOS 13.6 or later.

JoanMendoza, 2024-08-19:
On my ARM mac mini (M1) neither is working. 

The arm version gives me the same error that was posted in the pictures.

The x86 version allows me to uncompress it and open it. Once I to open it, there is an error that says it can't initialize somthing.  I don't know if this is a known error or if Rosetta is supposed to be able to run the x86 version of freecad. However I know that i was able to run the the x86 version successfully before switching to the arm version.  

[image]

smr02, 2024-08-19:
@BillTuttle It's the ARM package
however, the conda package `conda install freecad/label/dev::freecad` is working

artica1, 2024-08-19:
Same issue

Using `xattr -dr com.apple.quarantine /Applications/FreeCAD.app` lets the app open, but it fails with the same 'Failed to init from config' error that appears when running the x86 version.

wiz78, 2024-08-21:
FreeCAD_weekly-builds-38495-conda-macOS-arm64-py311 doesn't start on my mac either

luzpaz, 2024-08-21:
Attempts to fix are happening at #293

oursland, 2024-08-23:
The Intel build should be working again. The Apple Silicon build requires additional steps.  My notes on the investigation are [here](https://github.com/FreeCAD/FreeCAD-Bundle/pull/293#issuecomment-2306410478).

The QRD is that FreeCAD was launched by a shell script that caused macOS to think the application required Rosetta (it does not).  The shell script was replaced by a native program that fixes the Rosetta issue, but I had an error in it.  The fixed version works on Intel, but Gatekeeper (macOS code signature enforcement) behaves differently on Apple Silicon.  A command at the CLI will permit running the Apple Silicon version.  A better solution that does not require user intervention is being investigated.

pheraph, 2024-08-23:
I can confirm that the following steps fix the problem for me:

1. Download and install: https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-38495-conda-macOS-arm64-py311.dmg
2. Remove the quarantine attribute: `sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app`
3. Run FreeCAD

maxwxyz, 2024-08-23:
this was always the case for me in the weekly builds for over a year now :D
Just right click and then click open.

pheraph, 2024-08-23:
> this was always the case for me in the weekly builds for over a year now :D Just right click and then click open.

The whole point of this issue is that right clicking isn’t doing the job. 🫠

oursland, 2024-08-23:
My hypothesis is that Apple has decided to change the behavior of unsigned builds that are identified as Intel vs ARM.  The Intel builds permit right-click and open to run unsigned code from the internet, but ARM builds do not.

Before the recent changes to the FreeCAD launcher, the launcher was a architecture-independent shell script.  Apple's Gatekeeper could not identify the application as an ARM binary, which is why it claimed to require Rosetta 2 despite not needing it.  This also meant that the legacy behavior of right-click and open would work.  Now that the ARM binaries are correctly identified, they will not open without removing the attribute.

skandragon, 2024-08-24:
> My hypothesis is that Apple has decided to change the behavior of unsigned builds that are identified as Intel vs ARM. The Intel builds permit right-click and open to run unsigned code from the internet, but ARM builds do not.

I'm not sure about this.  I've used right-click to open on other projects before and it works.  I'm not about to say Apple would not force this issue by expiration of old systems, but I don't know why other downloads would allow me to right-click and open (and they are ARM64 builds, I have checked) while FreeCAD is some special case.

oursland, 2024-08-24:
@skandragon See my analysis here: https://github.com/FreeCAD/FreeCAD-Bundle/pull/293#issuecomment-2306410478

theosib, 2024-08-27:
Same problem with FreeCAD_weekly-builds-38553-conda-macOS-arm64-py311.

oursland, 2024-08-27:
@theosib see my comment here: https://github.com/FreeCAD/FreeCAD-Bundle/pull/293#issuecomment-2312765494

pilhuhn, 2024-09-11:
Same for me with the rc1 build on macos/arm

wblondel, 2024-09-11:
I had the same issue with the 1.0RC1 arm64 version for MacOS, however this allowed me to run it:
`xattr -dr com.apple.quarantine "/Applications/FreeCAD.app"`

JohnOCFII, 2024-09-13:
Today's signed bundle launches correctly on my M1/arm Mac.

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.28
  * Curves 0.6.47
```

hyarion, 2024-09-28:
Today's signed bundle does not launch correctly on my M1/arm Mac `<string>1.1.0dev-38843</string>`
`sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app` makes it work.
[image]

JohnOCFII, 2024-09-28:
> Today's signed bundle does not launch correctly on my M1/arm Mac `<string>1.1.0dev-38843</string>`
> `sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app` makes it work.

As far as I know, the current plan is the Release Candidates and final versions will be signed, but the weekly development builds won't be signed.

Yes - known issue.  I believe that'll be resolved when @oursland and/or @looooo refine the build process.

rolfeisenhut, 2024-10-14:
I have the same issue with the weekly build, because it was not codesigned. 
It could be fixed by code signing with locale codesigning tool after expanding zip file , b ut before first time start it.
Then start the application and confirm in security to run this application. 
After that it works fine, but must be resigned and confirmed after each new weekly version.
This is very annoying - so i vote for code signing the weekly  arm64 version as well.

Rolf

martin1951, 2024-10-26:
Same problem on my M2 MacBook Pro running Sonoma 14.6.1. I don't have the problem with the pre-release FreeCAD V1.0.

oursland, 2024-10-27:
@martin1951 Run `xattr -r -d com.apple.quarantine /Applications/FreeCAD.app` to remove the quarantine attribute from the app that prevents macOS from running programs downloaded from the internet.

psommerfeld, 2024-12-21:
There are multiple tickets about the weekly ARM Mac builds not running because of the xattr issue, and I think this issue will keep tripping Mac users up and will keep being reported. I see at least 5 others with a short search:
https://github.com/FreeCAD/FreeCAD-Bundle/pull/290
https://github.com/FreeCAD/FreeCAD-Bundle/pull/293
https://github.com/FreeCAD/FreeCAD-Bundle/issues/329
https://github.com/FreeCAD/FreeCAD/issues/17853
https://github.com/FreeCAD/FreeCAD/issues/16091

If signing the weekly builds is too burdensome, at least the Weekly Build page (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) can be updated to instruct Mac users:

### How-to use

1. Download the appropriate asset for your OS
2. Unpack the bundle to any folder on your system
3. Launch the application
- **Windows**
Run \bin\FreeCAD.exe in the extracted directory
- **macOS**
_**>>> In a Terminal window, clear attributes that will prevent it from running: `xattr -c /Applications/FreeCAD.app` <<<**_
Launch/FreeCAD.app in the extracted directory
- **Linux**
Open the *.AppImage

martinsp, 2025-01-12:
Here are instructions on how to install & use Apple signing certificate in GitHub Actions: https://docs.github.com/en/actions/use-cases-and-examples/deploying/installing-an-apple-certificate-on-macos-runners-for-xcode-development

This could be used to automatically sign the weekly build and avoid the issue with "damaged" app bundle

KeithSloan, 2025-03-26:
I tried xattr -cr /Applications/FreeCAD.app

On starting FreeCAD I briefly see the Startup screen but then nothing. Mac is MacBook Air - M3

KeithSloan, 2025-03-26:
Starting from a terminal
```
/Applications/FreeCAD.app*/C*/M*/FreeCAD   
FONTCONFIG_FILE=/etc/fonts/fonts.conf
FONTCONFIG_PATH=/etc/fonts
GIT_SSL_CAINFO=/Applications/FreeCAD.app/Contents/Resources/ssl/cacert.pem
HOME=/Users/ksloan
HOMEBREW_CELLAR=/opt/homebrew/Cellar
HOMEBREW_PREFIX=/opt/homebrew
HOMEBREW_REPOSITORY=/opt/homebrew
INFOPATH=/opt/homebrew/share/info:
LANG=UTF-8
LD_LIBRARY_PATH=/Applications/FreeCAD.app/Contents/Resources/lib
LOGNAME=ksloan
LaunchInstanceID=6E7C7F01-A917-49C0-99E5-0D5D11EB9741
OLDPWD=/Users/ksloan/github
PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/Apple/usr/bin
PREFIX=/Applications/FreeCAD.app/Contents/Resources
PWD=/Users/ksloan/github/FreeCAD_GDML_Test_Collection
PYTHONHOME=/Applications/FreeCAD.app/Contents/Resources
PYTHONPATH=/Applications/FreeCAD.app/Contents/Resources
QT_MAC_WANTS_LAYER=1
SECURITYSESSIONID=186a3
SHELL=/bin/zsh
SHLVL=1
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.hfGOFE7YPb/Listeners
SSL_CERT_FILE=/Applications/FreeCAD.app/Contents/Resources/ssl/cacert.pem
TERM=xterm-256color
TERM_PROGRAM=Apple_Terminal
TERM_PROGRAM_VERSION=455
TERM_SESSION_ID=75A4C115-7F69-4F83-B763-8AC0F1983506
TMPDIR=/var/folders/31/h9n3pl253yz7s27kcrh362340000gn/T/
USER=ksloan
XPC_FLAGS=0x0
XPC_SERVICE_NAME=0
_=/Applications/FreeCAD.app/Contents/MacOS/FreeCAD
__CFBundleIdentifier=com.apple.Terminal
Running: /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Error: Failed to open library "/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib"! Error: dlopen(/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib, 0x0005): tried: '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file, not in dyld cache)!
FreeCAD 1.1.0, Libs: 1.1.0devR40655 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Error in InitApplications embedded null byte
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 707, in <module>
  File "<string>", line 98, in InitApplications
  File "<frozen posixpath>", line 416, in realpath
  File "<frozen posixpath>", line 451, in _joinrealpath
ValueError: embedded null byte
--------------------------------------------------------------------------------
(qt.qpa.fonts) Populating font family aliases took 105 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
Traceback (most recent call last):
  File "<string>", line 303, in <module>
  File "<string>", line 133, in InitApplications
<class 'AttributeError'>: module 'FreeCAD' has no attribute '__ModDirs__'
Error in FreeCADGuiInit.py: module 'FreeCAD' has no attribute '__ModDirs__'
2025-03-26 17:07:17.898 freecad[31051:4628561] +[IMKClient subclass]: chose IMKClient_Modern
2025-03-26 17:07:17.898 freecad[31051:4628561] +[IMKInputSession subclass]: chose IMKInputSession_Modern
ksloan@Mac FreeCAD_GDML_Test_Collection % 
```

JohnOCFII, 2025-03-26:
> Error: Failed to open library 
> FreeCAD 1.1.0, Libs: 1.1.0devR40655 (Git)

@KeithSloan - `40655` had additional issues.  The latest weekly build `40971`, available here, should work fine after executing the `xattr` command.

https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

KeithSloan, 2025-03-26:
> [@KeithSloan](https://github.com/KeithSloan) - `40655` had additional issues. The latest weekly build `40971`, available here, should work fine after executing the `xattr` command. [...]

Thanks now fires up, barfs about not finding PySide2, where do I raise issues development builds?

JohnOCFII, 2025-03-26:
> Thanks now fires up, barfs about not finding PySide2, where do I raise issues development builds?

This link is the correct place for FreeCAD issues: https://github.com/FreeCAD/FreeCAD/issues