FreeCAD/FreeCAD-Bundle #294: Is FreeCAD (0.22 dev, build 38495) dropping macOS Monterey compatibility?

pgilfernandez, 2024-08-20:
Maybe it's a dev decision but it sound strange to me...
Tested with `FreeCAD-38495-conda-macOS-x86_64-py311`

Cheers

---

OS: macOS 12.6.7

longrackslabs, 2024-08-20:
still a problem with buiild 38495.  I hope this is just a bug introduced in the last weeks and not a product decision.

[image]

pgilfernandez, 2024-08-20:
Well, in my case if I double click it, it says the following:
[image]

wiz78, 2024-08-21:
I think it's broken. FreeCAD_weekly-builds-38495-conda-macOS-arm64-py311 doesn't work on Sonoma either: it always complains that it's corrupted and cannot be started.

luzpaz, 2024-08-21:
Likely being addressed in #293

oursland, 2024-08-23:
The Intel build should be working again. The Apple Silicon build requires additional steps.  My notes on the investigation are [here](https://github.com/FreeCAD/FreeCAD-Bundle/pull/293#issuecomment-2306410478).

The QRD is that FreeCAD was launched by a shell script that caused macOS to think the application required Rosetta (it does not).  The shell script was replaced by a native program that fixes the Rosetta issue, but I had an error in it.  The fixed version works on Intel, but Gatekeeper (macOS code signature enforcement) behaves differently on Apple Silicon.  A command at the CLI will permit running the Apple Silicon version.  A better solution that does not require user intervention is being investigated.

pgilfernandez, 2024-08-23:
Great @oursland!
We'll have to wait for next weekly builds in order to test it.

Thanks!

pgilfernandez, 2024-08-24:
Build 38506 still can't be opened in macOS Monterey...

JohnOCFII, 2024-08-24:
> Build 38506 still can't be opened in macOS Monterey...

Intel processor or Arm (M1/M2/M3) processor?  The FreeCAD build for arm is known to still be broken.

pgilfernandez, 2024-08-24:
Intel, I tried the Intel one

oursland, 2024-08-24:
@pgilfernandez Did you open `Finder`, then right mouse click on `FreeCAD.app` and click `Open`?

pgilfernandez, 2024-08-24:
No, I didn't... I just opened the DMG file and checked that the icon had a "forbidden" sign and didn't go further...
I will not be able to check it again in 10 days as I went out on a trip. Maybe someone else can kindly check it.
I'll stay tuned and be glad to try it again in 10 days.

longrackslabs, 2024-08-24:
build 506 still broken for me on on monterrey.  x86

oursland, 2024-08-25:
@longrackslabs 

Can you open a terminal and run `wget https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-38506-conda-macOS-x86_64-py311.dmg` to download the `.dmg` and attempt to install that way?

oursland, 2024-08-25:
I have been scrambling to figure out where this minimum version was stated, because it wasn't in the `Info.plist` file within the `FreeCAD.app` as is typically the case.

The answer appears to be the compiler used to build the source as determined by `otool -l /Applications/FreeCAD.app/Contents/MacOS/FreeCAD`:

    ...
    Load command 10
          cmd LC_BUILD_VERSION
      cmdsize 32
     platform 1
        minos 13.6
          sdk 14.2
       ntools 1
         tool 3
      version 1022.1
    ...

This appears to be another side-effect of compiling the launcher instead of the shell script which broke macOS's ability to analyze the `FreeCAD.app`.

Recently, the macOS `x86_64` runner was moved to `macos-13` which I believe is resulting in this minimum OS version.  I'll submit a PR to move it back to `macos-12` to see if that reduces the minimum version to one that supports older versions of macOS.

In the future, I am interested to see if the `macos-latest` can be set to build for older versions of macOS to take advantage of the faster runners to build for all platforms.

oursland, 2024-08-25:
PR https://github.com/FreeCAD/FreeCAD-Bundle/pull/296 has been submitted to address the version compatibility issue.

satiromarra, 2024-08-26:
@oursland  I had problems with all the dmg downloaded through safari.
I also downloaded the last one (build 38506 for arm macOS) on safari and I had problems, but when I did it with `wget` it get off and I was able to open it without problems.
For now, it works correctly.

oursland, 2024-08-26:
@pgilfernandez This issue should be resolved.  When you have a chance, please test the weekly and see if it installs and runs on macOS Monterey.

pgilfernandez, 2024-08-26:
> @pgilfernandez This issue should be resolved.  When you have a chance, please test the weekly and see if it installs and runs on macOS Monterey.

Indeed.
By the way, if somebody could verify it before next Monday that is when I'll be able to test it by myself we could resolve it and close the issue...

longrackslabs, 2024-08-27:
Build 38553 installs and runs on Monterrey (12.7.6) again!  Nice work and thanks for fixing.

pgilfernandez, 2024-08-27:
Great!
I'm closing the issue then.

Thank to everybody

closed 2024-08-27 by pgilfernandez