FreeCAD/FreeCAD-Bundle #299: Add github issue template to curb the amount of tickets that are meant for FreeCAD/FreeCAD

luzpaz, 2024-08-26:
@maxwxyz this might be the better solution because people are still ignoring the pinned issue

maxwxyz, 2024-08-27:
Yes but I think they will still ignore it, same as the checkbox 'I have searched for similar issues'

luzpaz, 2024-08-27:
I'll make a PR and we can discuss strategies

JohnOCFII, 2024-08-27:
With people blindly following links -- some that go back and forth between the `FreeCAD` and `FreeCAD-Bundle` repositories -- anything to clearly show that **THIS** isn't the main FreeCAD repository would be useful.

maxwxyz, 2024-08-27:
What if we do not allow issues here and instead the packaging related issues are reported in the main repo? I mean there are already some with the tag.

luzpaz, 2024-08-27:
Or the opposite, move all the packaging issues here ?

JohnOCFII, 2024-08-27:
> Or the opposite, move all the packaging issues here ?

Or -- does this need to be a separate repo?

luzpaz, 2024-08-27:
> Or -- does this need to be a separate repo?  

Separate from what? This repo is dedicated to building and packaging FC. What would you propose for a new repo that is different than this one ?

JohnOCFII, 2024-08-28:
> Edit: oh, are you proposing merging this repo back into FreeCAD/FreeCAD ?

Yes, that was my question -- is the hassle with issues showing up in the wrong respository enough of a reason to try to handle the current `FreeCAD-Bundle` Repository activities as just part of the main FreeCAD repository.

And it was truly at the question stage -- I don't know near enough about the mechanics to know if this is a bad idea or not.

commit 6ce41281 (2024-08-27): Add bug report issue template (#299)

Added another level of abstraction against incorrect filing of bugs.  
Closes #299
PR #301 (2024-08-28): Add bug report issue template (#299)
Added another level of abstraction against incorrect filing of bugs.  
Closes #299
Preview: https://github.com/FreeCAD/FreeCAD-Bundle/blob/issue-template/.github/ISSUE_TEMPLATE/bug-report.md
closed 2024-08-28 by chennes