FreeCAD/FreeCAD-Bundle #302: Very long loading times of some Themes in AppImage

MiqCG, 2024-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When in Light theme, shortcut for new file (Control + n) and others (Alt + f , Alt + e) don't work. If new file is created throw File menu with the mouse, then they work (see indications in the bottom part of the screencast).
I think there is the same problem with revision 37841 and 37543.
In some situations, clicking with the mouse on the menu bar doesn't work either.
On the other hand, looks like loading/activating the Light theme is much much slower than Classic theme, as you can see in the attached video.

I have searched in Github for [is:issue light theme shortcut] and [is:issue light theme new file] with no relevant results

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git) AppImage
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * sheetmetal 0.4.20
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.24
```

### Subproject(s) affected?

Core

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/18265355/c728a522-ab91-4f0b-9fd5-fb72ff43206a

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-26:
This is a first impression UX issue and frankly kind of embarrassing if we release 1.0 with it still open. The wait time is misleading and people think FC has crashed since there is no visual indication that the operation is ongoing.

luzpaz, 2024-09-26:
@probonopd any chance you could help troubleshoot the delay ?

MisterMakerNL, 2024-09-26:
Can we check if it is the preference pack or the stylesheet. And if the stylesheet, can we check if it is the overlay stylesheet or the normal one or maybe both?

Syres916, 2024-09-27:
@MisterMakerNL this testing was carried out using build 38827 AppImage, extracting and then repacking the AppImage after each change to the `FreeCAD Dark.cfg` to only load what I required at each stage. Here are my findings and to be honest as I expected:

<!DOCTYPE html>

<html>
<head>
	
</head>

<body>

Change Requested | Time To complete Change
-- | --
Preference Pack Only | 4.02 seconds
Pref Pack + Stylesheet Only | 16.36 seconds
Pref Pack + Overlay Only | 6.45 seconds
Pref Pack + Stylesheet & Overlay | 33.75 seconds

</body>

</html>

MisterMakerNL, 2024-09-27:
Do you check if the latest stylesheet are installed, or are you doing a fresh install? I have had issues where it didn't overwrite the older stylesheets?

Syres916, 2024-09-27:
My testing was with no existing user.cfg/system.cfg/FreeCAD.conf config files and only using the stylesheets provided in the AppImage, no other stylesheets/addons were involved. It's quite clear that the more lines of text in any of the cfg or qss files makes the changeover longer, as simple as that IMHO. It's basically 5.5 times slower as a packed AppImage versus an extracted one.

MisterMakerNL, 2024-09-27:
But we are talking about files less then 80Kb that should needn't need 30 second load time...
Normally how I trouble should these issues is just delete half of the stylesheet, then test. if still long load times delete again half, and so keep going until I found the issue. 
This is how I last time found Qwidget be the cause of the loading issues.

probonopd, 2024-09-28:
Does this happen only on KDE Plasma on Wayland? Then it might be related to https://github.com/AppImage/AppImageKit/issues/1351, the root cause of which is still unknown.

Can it be confirmed that the slowdown doesn't happen if one runs with `--appimage-extract-and-run`?

Syres916, 2024-09-28:
> Does this happen only on KDE Plasma on Wayland?

Nope, I used `OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)` for the above testing

> Can it be confirmed that the slowdown doesn't happen if one runs with `--appimage-extract-and-run`?

I'll report back when my box is freed up from compiling.

Syres916, 2024-09-28:
@probonopd using `--appimage-extract-and-run` it takes 7.4 seconds instead of 33.75 seconds

probonopd, 2024-09-28:
Is this on Wayland? If so, can you try on Xorg?

Syres916, 2024-09-28:
I thought the information above was clear that it's not on Wayland it's X-Cinnamon

probonopd, 2024-09-28:
OK, then I don't have a hypothesis at the moment. Might be worth to experiment with different compression algoritthms (we want to standardize on zstandard) and compression levels.

kinghat, 2024-09-28:
> Does this happen only on KDE Plasma on Wayland? Then it might be related to [AppImage/AppImageKit#1351](https://github.com/AppImage/AppImageKit/issues/1351), the root cause of which is still unknown.
> 
> Can it be confirmed that the slowdown doesn't happen if one runs with `--appimage-extract-and-run`?

think this is fixed here: https://bugs.kde.org/show_bug.cgi?id=492584 my freecad freezing issues are not this one. also not sold on themes being the issue either.

adrianinsaval, 2024-11-02:
is this still relevant?

kinghat, 2024-11-02:
> is this still relevant?

i no longer have freezing/loading issues with the appimage on:

> Operating System: Fedora Linux 41
KDE Plasma Version: 6.2.2
KDE Frameworks Version: 6.7.0
Qt Version: 6.8.0
Kernel Version: 6.11.5-300.fc41.x86_64 (64-bit)
Graphics Platform: Wayland

MiqCG, 2024-11-02:
I think it is still relevant but you can judge if the times are correct. 
In a fresh install of 39092 revision ([as 39100 is problematic](https://github.com/FreeCAD/FreeCAD/issues/17618)) the loading times in the start page are around 1 second or less. But if you choose one theme and then close FreeCAD and reopen it, the theme change times  through Preferences are:
- from Light to Dark ~ 20-25 sec
- from Dark to Light ~ 20-25 sec 
- from Light to Classic < 5 sec
- from Dark to Classic < 5 sec
- from Classic to Dark ~ 15 sec
- from Classic to Light ~ 15 sec

Isn't curious the time difference?

```
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39092 (Git) AppImage
Build type: Release
Branch: main
Hash: a65d05c6e495038168eefcba6a1e43218fbb3a20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

probonopd, 2024-11-03:
Maybe it would help to check with `strace` what happens when it needs ~20-25 seconds. Is it trying to access some files then?

MiqCG, 2024-11-03:
> Maybe it would help to check with `strace` what happens when it needs ~20-25 seconds. Is it trying to access some files then?

Didn't know `strace` tool.
I have made:
`strace ./FreeCAD_weekly-builds-39092-conda-Linux-x86_64-py311.AppImage`
and there is no output after FreeCAD starting: no output while changing themes neither when opening a file.
Am I doing it correctly?

probonopd, 2024-11-03:
You will probably need `strace -f -eopen <...>` and possibly `sudo -e` in front of that.

MiqCG, 2024-11-03:
I have made 
`sudo strace -f -eopen ./FreeCAD_weekly-builds-39092-conda-Linux-x86_64-py311.AppImage `
and the output has been different than previously but without output during or after the long theme changes.

On the other hand
`sudo -e strace -f -eopen ./FreeCAD_weekly-builds-39092-conda-Linux-x86_64-py311.AppImage `
and
`strace -f -eopen ./FreeCAD_weekly-builds-39092-conda-Linux-x86_64-py311.AppImage` 
gave me errors and didn't start FreeCAD.

benj5378, 2024-11-10:
I'm not sure if this is a bundling issue tbh. Seems like a coding issue: I have noticed it's only slow when choosing in Preference manager and not when choosing on setup start page.

ChrisJones79, 2024-11-10:
Very long pauses (stalled, unresponsive) between any UI interactions. Open AppImage, chose Dark Theme, clicked New Parametric Part... 5 - 10 seconds of unresponsiveness.

```
OS: Ubuntu 24.10 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39153 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

kinghat, 2024-11-10:
> Very long pauses (stalled, unresponsive) between any UI interactions. Open AppImage, chose Dark Theme, clicked New Parametric Part... 5 - 10 seconds of unresponsiveness.
>
> ```
> OS: Ubuntu 24.10 [...]

what plasma version? you need to be on 6.2 or higher iirc.