FreeCAD/FreeCAD-Bundle #305: 1.0RC is being released as 1.0.0 and this will cause confusion.

luzpaz, 2024-09-11:
[image]

Is this due to some technical limitation or just an oversight ?

TIA for your attention @adrianinsaval

oursland, 2024-09-11:
The issue is narrowed down to how [`get_freecad_version.py`](https://github.com/FreeCAD/FreeCAD-Bundle/blob/main/conda/scripts/get_freecad_version.py) determines the version information.

oursland, 2024-09-11:
@chennes Is this supposed to be `FreeCAD_1.0.0RC1-...` or `FreeCAD_1.0.0_RC1-...`?

chennes, 2024-09-11:
I don't have a preference.

oursland, 2024-09-11:
I'll follow the convention in the About box:

[image]

`FreeCAD_1.0.0RC1-...`

closed 2024-09-13 by adrianinsaval