FreeCAD/FreeCAD-Bundle #307: Due to RC release perhaps publishing nightlies will help get us to 1.0 faster ?

luzpaz, 2024-09-11:
I've been thinking about this a bunch and finally i saw [this comment](https://github.com/FreeCAD/FreeCAD-Bundle/issues/304#issuecomment-2344721092) and decided to make the request.

We could either make a nightly tag or just increase the frequency of the weekly tag to publish builds every day ?

adrianinsaval, 2024-09-12:
ugh I should have answered here. Made some long answers on the other issue

@oursland probably a better idea to continue that conversation here?

oursland, 2024-09-12:
@adrianinsaval 's [response](https://github.com/FreeCAD/FreeCAD-Bundle/issues/304#issuecomment-2345075515) explains that doing so would block the CI runners for 2.5 hours (or longer) and prevent regular CI tasks.

closed 2024-10-27 by adrianinsaval