FreeCAD/FreeCAD-Bundle #308: [Problem] MacOS Arm version appears to be corrupted/damaged

eggy111, 2024-09-13:
**Describe the bug**
No matter how many times I re-download the file it comes out corrupted or damaged.  Let me know if I am missing anything, or can provide anything else

**To Reproduce**
Steps to reproduce the behavior:
1. Download the MacOS ARM version
2. Open the .dmg file
3. Move the .app to Applications
4. Run the program and it says it is corrupted, or damaged

adrianinsaval, 2024-09-13:
duplicate of #291, this happens when the binaries are not signed. In any case I'm reuploading a signed version soon so the problem will go away for RC1 (and actual release will also be signed) or you can circumvent this issue with
```
xattr -r -d com.apple.quarantine FreeCAD.app
```

closed 2024-09-13 by adrianinsaval

adrianinsaval, 2024-09-13:
signed bundle is up, please test

JohnOCFII, 2024-09-13:
> signed bundle is up, please test

Signed bundle launches correctly on my M1/arm Mac.

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.28
  * Curves 0.6.47
```