FreeCAD/FreeCAD-Bundle #310: [Problem] Question about weeklies version 1.1.0 vs 1.0-RC1

leoheck, 2024-09-15:
AppImage Weekley says `1.1.0-dev`. Shouldn't it be `1.0.RC2` since `1.0` was not released yet?
Or if it was released, where do they comunicate this?

```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt defaul
```

luzpaz, 2024-09-15:
My take: dev is now renamed to 1.1.0dev.xxxxx  
RCs will either get committed to directly for fixing blockers and other minor issues OR cherrypick commits from dev version. This will happen until 1.0 will be released.

leoheck, 2024-09-15:
Hm, alright, thanks dude. Let me see if someone else has something to say and then I will close this issue by Monday.

adrianinsaval, 2024-09-18:
1.0 is a separate branch, main will be the base for 1.1 so it is now 1.1dev. In any case please don't create issues just for asking questions, there are more appropriate places for that, it can be the forum or a discussion topic on github

closed 2024-09-18 by adrianinsaval