FreeCAD/FreeCAD-Bundle #312: Compile from source to opensuse tumbleweed

tckosvic, 2024-09-18:
compile source to opensuse tumbleweed from config throws error below;

`Make Error at /usr/lib64/cmake/Boost-1.86.0/BoostConfig.cmake:141 (find_package):
  Found package configuration file:

    /usr/lib64/cmake/boost_python-1.86.0/boost_python-config.cmake

  but it set boost_python_FOUND to FALSE so package "boost_python" is
  considered to be NOT FOUND.  Reason given by package:

  No suitable build variant has been found.

  The following variants have been tried and rejected:

  * libboost_python-py3.so.1.86.0 (3.11, Boost_PYTHON_VERSION=3.10)

Call Stack (most recent call first):
  /usr/lib64/cmake/Boost-1.86.0/BoostConfig.cmake:262 (boost_find_component)
  /usr/share/cmake/Modules/FindBoost.cmake:610 (find_package)
  src/Mod/MeshPart/App/CMakeLists.txt:131 (find_package)
`
In stalled boost_python modules are:
`(base) tom@localhost:~> sudo zypper search libboost_python
[sudo] password for root: 
Loading repository data...
Reading installed packages...

S  | Name                                 | Summary                                      | Type
---+--------------------------------------+----------------------------------------------+--------
i  | libboost_python-py3-1_86_0           | Boost.Python runtime library                 | package
   | libboost_python-py3-1_86_0-32bit     | Boost.Python runtime library                 | package
i  | libboost_python-py3-1_86_0-devel     | Development headers for Boost.Python library | package
i+ | libboost_python-py3-1_86_0-x86-64-v3 | Boost.Python runtime library                 | package
i+ | libboost_python3-devel               | Development headers for Boost.Python library | package
(base) tom@localhost:~> 
`
thanks for any ideas,      tom kosvic

luzpaz, 2024-09-18:
Where are you pulling source from ?

It should be from: https://github.com/FreeCAD/FreeCAD/

luzpaz, 2024-10-23:
@tckosvic ping

adrianinsaval, 2024-10-27:
not the place to ask this, open a forum thread instead: https://forum.freecad.org/viewforum.php?f=4&sid=14540ab38d0f9aaea19718191d812f1c

closed 2024-10-27 by adrianinsaval