FreeCAD/FreeCAD-Bundle #314: FreeCAD conda install not compatible with ifcopenshell 0.8.0

smr02, 2024-10-03:
I'm trying to install a current version of IfcOpenShell together with FreeCAD in a conda environment. 
The latest compatible version seems to be `ifcopenshell==v0.7.0.231127=py311_all_h8f4786a_202`. The latest version is 0.8.0

Here is my .yml file: 

``` test_env.yml

name: test_env
channels: 
  - freecad/label/dev
  - conda-forge
  - defaults
dependencies:
  - freecad/label/dev::freecad
  - ifcopenshell==0.8.0
```

``` 
conda env create -f test_env.yml --dry-run
Channels:
 - freecad/label/dev
 - conda-forge
 - defaults
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package freecad-1.0rc1-py311h5536ab1_0 requires vtk-base >=9.2.6,<9.2.7.0a0, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ freecad is installable with the potential options
│  ├─ freecad [1.0rc1|1.0rc2|1.1dev] would require
│  │  ├─ coin3d >=4.0.3,<4.1.0a0 , which requires
│  │  │  └─ libexpat >=2.6.3,<3.0a0 , which can be installed;
│  │  └─ vtk-base >=9.2.6,<9.2.7.0a0  with the potential options
│  │     ├─ vtk-base 9.2.6 would require
│  │     │  └─ libzlib >=1.2.13,<1.3.0a0 , which can be installed;
│  │     └─ vtk-base 9.2.6 would require
│  │        └─ libexpat [<2.6 |>=2.5.0,<2.6.0a0 ], which conflicts with any installable versions previously reported;
│  └─ freecad 1.1dev would require
│     ├─ coin3d >=4.0.3,<4.1.0a0 , which can be installed (as previously explained);
│     ├─ libboost >=1.82.0,<1.83.0a0  with the potential options
│     │  ├─ libboost 1.82.0 would require
│     │  │  └─ icu >=72.1,<73.0a0 , which can be installed;
│     │  └─ libboost 1.82.0, which can be installed;
│     ├─ qt-main >=5.15.13,<5.16.0a0 , which requires
│     │  └─ icu >=73.2,<74.0a0 , which conflicts with any installable versions previously reported;
│     └─ vtk-base >=9.2.6,<9.2.7.0a0  with the potential options
│        ├─ vtk-base 9.2.6, which can be installed (as previously explained);
│        └─ vtk-base 9.2.6, which cannot be installed (as previously explained);
└─ ifcopenshell 0.8.0  is not installable because there are no viable options
   ├─ ifcopenshell 0.8.0 would require
   │  └─ libzlib >=1.3.1,<2.0a0 , which conflicts with any installable versions previously reported;
   ├─ ifcopenshell 0.8.0 would require
   │  ├─ libboost >=1.86.0,<1.87.0a0  with the potential options
   │  │  ├─ libboost 1.86.0 would require
   │  │  │  └─ icu >=75.1,<76.0a0 , which can be installed;
   │  │  └─ libboost 1.86.0 conflicts with any installable versions previously reported;
   │  └─ libzlib >=1.3.1,<2.0a0 , which conflicts with any installable versions previously reported;
   └─ ifcopenshell 0.8.0 would require
      └─ python >=3.13.0rc2,<3.14.0a0 , which requires
         └─ _python_rc, which does not exist (perhaps a missing channel).
```

luzpaz, 2024-10-17:
@adrianinsaval any idea?

oursland, 2024-10-18:
`freecad/label/dev::freecad` depends upon `ifcopenshell` at build time.  Consequently, the version dependency is pinned to that version or a compatible one (newer patch version).  It looks like the dependency compatibility is limited by incompatibilities in `libzlib` which is required by `vtk-base` at `1.2.13` and `ifcopenshell` at `1.3.1`.

In this instance the version of `ifcopenshell` that was used to build was `v0.7.0` as seen here: https://github.com/FreeCAD/FreeCAD-Bundle/actions/runs/11336342929/job/31526105633#step:9:363

To resolve this, `vtk-base` would need to be rebuilt against `libzlib` at `1.3.1`.  However, it's possible that this version of `vtk-base` was selected to be compatible with another dependency.

furgo16, 2025-02-10:
See https://github.com/FreeCAD/FreeCAD/issues/19051#issuecomment-2649221339 (please feel free to correct if necessary).