FreeCAD/FreeCAD-Bundle #315: Unexpected installation failure due to low available space on Windows main drive

hgroover, 2024-10-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This was a slightly puzzling issue I encountered installing 1.0 rc2 on Windows 10 Pro, on a Xeon machine. I had under 1gb free on my C drive before downloading the installer to the default Downloads folder (which further reduced available space on C: to around 500mb).

I selected a folder on my additional drive (F:) and extraction started up to a certain point, where it failed with an error writing some file (on two separate attempts these were different files). I rebooted before the second attempt because after two failures I could not start the installer - it complained of failing an integrity check. I also compared the sha256 hash and it matched the expected one provided in the artifact download folder.

After freeing up some gigabytes on the C drive I attempted installation again (still installing to F: where I have over 500gb free) and it succeeded. My speculation is that there is some function in un-archiving and extracting files that uses temp space, and that got exhausted in doing the full install, causing the failure. 

This is not an actual defect in the installer (PEBKAC) but in retrospect it was an issue I should have thought to address. Uninstalling my two-versions-ago previous and three-versions-ago previous Freecad versions freed up quite a bit of space. Perhaps a cautionary note that the Windows default drive should also have a minimum 2gb free space (or something) would be helpful.

Failed installation (from a second attempt) attached.
[install-failure1.log](https://github.com/user-attachments/files/17256296/install-failure1.log)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * A2plus 0.4.65
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-17:
Any proposal to avoid this usecase ?

oursland, 2024-10-18:
It definitely appears in the logfile that the C: drive is being used for temporary storage.

It looks like the provided artifact is a `.7z` archive.  Am I correct in understanding that this log is generated by the extracting program such as 7zip?

adrianinsaval, 2024-10-27:
> It definitely appears in the logfile that the C: drive is being used for temporary storage.
> 
> It looks like the provided artifact is a `.7z` archive. Am I correct in understanding that this log is generated by the extracting program such as 7zip?

the installer is an .exe generated using nsis and it is configure to use lzma compression, nsis might be using 7z as their backend for compression I don't know for sure.

there aren't many references to C:\ in the log, it looks like nsis is putting a few files it needs itself on the TEMP directory (and I doubt these take 500mb) but the rest seem to be getting extracted to F:\ correctly
```
File: overwriteflag=0, allowskipfilesflag=2, name="icuin.dll"
File: wrote -2 to "F:\progs\Freecad 1.0\bin\icuin.dll"
Extract: error writing to file icuin.dll
Delete: DeleteFile("C:\Users\henry\AppData\Local\Temp\nsk9B2A.tmp\Banner.dll")
Delete: DeleteFile("C:\Users\henry\AppData\Local\Temp\nsk9B2A.tmp\modern-header.bmp")
Delete: DeleteFile("C:\Users\henry\AppData\Local\Temp\nsk9B2A.tmp\modern-wizard.bmp")
Delete: DeleteFile("C:\Users\henry\AppData\Local\Temp\nsk9B2A.tmp\nsDialogs.dll")
Delete: DeleteFile("C:\Users\henry\AppData\Local\Temp\nsk9B2A.tmp\nsProcess.dll")
Delete: DeleteFile("C:\Users\henry\AppData\Local\Temp\nsk9B2A.tmp\StartMenu.dll")
Delete: DeleteFile("C:\Users\henry\AppData\Local\Temp\nsk9B2A.tmp\System.dll")
Delete: DeleteFile("C:\Users\henry\AppData\Local\Temp\nsk9B2A.tmp\UserInfo.dll")
RMDir: RemoveDirectory("C:\Users\henry\AppData\Local\Temp\nsk9B2A.tmp\")
logging set to 1
Call: 17920
SetFlag: 13=6
```