FreeCAD/FreeCAD-Bundle #316: [Problem] AppImage updater should support unicode/emojis

luzpaz, 2024-10-05:
**Describe the bug**
When updating AppImages through the FreeCAD preferences, the dialog that comes up confirming their is an available update doesn't display emojis (that are shown the weekly build notes ([source](Weekly build notes: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds))

**To Reproduce**
Steps to reproduce the behavior:
1. Edit -> Preferences
2. AppImage
3. Click *Check for updates* button

**Result:** 
if you're on a slightly older weekly appimage then the dialog will display showing lack of support for emojis

**Screenshots**

[image]

Weekly build notes: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
[image]

**Additional context**
Not sure is if this is FreeCAD software related or AppImage software related.

adrianinsaval, 2024-10-05:
Instead I would remove emojis from the release page, this code is currently hosted at https://github.com/looooo/freecad.appimage_updater, probably should be moved to an org repo

luzpaz, 2024-10-18:
The point of that emoji is to get the attention of users who are text blind when quickly scanning the page.