FreeCAD/FreeCAD-Bundle #318: Python

13061951, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I start freecad from the weekly builds. I delete the previous version. On starting the latest Version, which is 38871. I get a dialog box that says.
No python @ c:user\downloads\freecad_weekly_builds 38728 conda windows x86_64_py311\freecad weekly builds-32728 conda-windows-x86-64-py311\bin\pythonW.exe.
I have to rest the dialogue box 4 times for the programme to run. After this everything seems alright. 
This also happens when I open version 21.1? 

### Full version info

```shell
When I start freecad from the weekly builds. I delete the previous version. On starting the latest Version, which is 38871. I get a dialog box that says.
No python @ c:user\downloads\freecad_weekly_builds 38728 conda windows x86_64_py311\freecad weekly builds-32728 conda-windows-x86-64-py311\bin\pythonW.exe.
I have to rest the dialogue box 4 times for the programme to run. After this everything seems alright. 
This also happens when I open version 21.1?
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-05:
This is a weird looking path:
`c:user\downloads\freecad_weekly_builds 38728 conda windows x86_64_py311\freecad weekly builds-32728 conda-windows-x86-64-py311\bin\pythonW.exe`

Can you also add your full [About](https://wiki.freecad.org/About) info please?

13061951, 2024-10-06:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * AirPlaneDesign 0.4.1
  * Assembly4 0.50.15
  * CurvedShapes 1.0.10
  * Curves 0.6.48
  * ExplodedAssembly
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Gridfinity 0.6.0
  * lattice2 1.0.0
  * MeshRemodel 1.10.25
  * POV-Ray-Rendering
  * Render 2024.9.25
  * sheetmetal 0.4.26
  * woodworking 0.21.2.33771

luzpaz, 2024-10-06:
Can you move the files to any drive beside c: ? (and try to run the executable ? )

13061951, 2024-10-08:
Good Morning, I have tried running this from an external drive and the same error occurs?

adrianinsaval, 2024-10-15:
can you open your config file at `%APPDATA%/FreeCAD/user.cfg` and search if that path is saved anywhere there and post what parameter holds this

13061951, 2024-10-15:
> DuckDuckGo removed one tracker. More<https://duckduckgo.com/-fndnWmaAhjGY9KbvM8v8VioefFhDChH6jYKRWsBSsg1lodhVKUzy7nZmb2_FgNkC20GOHJRnIwrNTMZFO70OwTBLdW8nmWBwARUPSdpb8IZ0P89-RFWaRJDMC-pcUJatAWBPh3YohJs9gdNKAqRzfP-91R-o75VP-kmGbvnk5K5hpWQz1m7ffCJ25LUQP-al91_ZLeIdV2XT0EjAHeqxNq2HaSoRa07NcItPoEoyNcPSxUnol3eHsv-H4qfXq-n30tVp0HANQjyUB4tvABsVBAHG6M5diVJttCNVTouAcnAA_oyJJQ8kwAJ-inH3nFDeW5ok5nX0aQcZawlLWpNFRVphZhKU8xJeS3wDOYxP-Li3Z18xmSsoRCkvedetFe1nZsN47U0rJ5Gxhd7tUQY77scEjJX5CO1LhtR7HERqVJ1oG8H_gaXSWsaK8BqIWSC1KYaNcvO5HA-sFZQPr3ryBJYVm-Y_yYg8GqpsVXJa5CCqmMNegA>
> Deactivate <https://duckduckgo.com/>

Thank you for getting back to me. I have Just returned from an operation in hospital, with further surgery next week. If I could get back to you when I feel able to I surely will.

Kindest Regards

Lynn Harris.

adrianinsaval, 2024-10-17:
I suspect the render workbench might have something to do with this