FreeCAD/FreeCAD-Bundle #322: Windows builds should have wrapper that invokes `bin\freecad.exe` instead of searching for it

luzpaz, 2024-10-20:
See https://github.com/FreeCAD/FreeCAD/issues/10099#issuecomment-1689057546

adrianinsaval, 2024-10-27:
wrong link?

luzpaz, 2024-11-23:
Yea, i can't locate the original ticket but it was a lot of explaining that needed to happen to help the user invoke the executable. 

By chance, i'm on a windows machine at the moment and tried install the Windows dev build and saw this in winzip app:

```
Error: Unable to create folder "C:\Users\foo\Documents\FreeCAD_weekly-builds-39287-conda-Windows-x86_64-py311\FreeCAD_weekly-builds-39287-conda-Windows-x86_64-py311\bin\Lib\site-packages\pkg_resources\tests\data\my-test-package_unpacked-egg\my_test_package-1.0-py3.7.egg\EGG-INFO"
Irrecoverable Error:  Extract operation failed
```