FreeCAD/FreeCAD-Bundle #326: Linux AppImage does not contain Wayland platform plugin

phrxmd, 2024-10-29:
**Describe the bug**
The Linux AppImage builds do not package the Qt Wayland platform plugin. As a result, the app launches only with X11 support in XWayland.

**To Reproduce**
1. Download AppImage from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
2. Launch from command line

**Observed behaviour**
```
> ./FreeCAD_weekly-builds-39092-conda-Linux-x86_64-py311.AppImage
FreeCAD 1.1.0, Libs: 1.1.0devR39092 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
Migrating Start Workbench to Start command... done.
The `.FCStd#` backup format is deprecated and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
```

**Expected behaviour**
FreeCAD should launch without complaining about not finding the Qt platform plugin for the graphics system it's being run on. The choice whether to run FreeCAD in Wayland or in X11 should be the user's.

luzpaz, 2024-10-31:
What's the process of including the "Qt Wayland platform plugin" in the AppImage ?

adrianinsaval, 2024-10-31:
There is a qt-wayland package in conda-forge however when I tried it in the past it broke the appimage, but that might have been due to package version missmatch, I have been working on upgrading the qt5 packages in conda-forge and hopefully it will work with that.

penguin-wrangler, 2024-11-04:
> There is a qt-wayland package in conda-forge however when I tried it in the past it broke the appimage, but that might have been due to package version missmatch, I have been working on upgrading the qt5 packages in conda-forge and hopefully it will work with that.

This broke the package. 39057 works (falling back to Xwayland due to missing platform plugin) while 39100 just crashes at startup:

xkbcommon: ERROR: failed to add default include path /home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/X11/xkb
QOpenGLFunctions created with non-current context
Program received signal SIGSEGV, Segmentation fault.

kinghat, 2024-11-05:
```
$ ./FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.1.0, Libs: 1.1.0devR39127 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

xkbcommon: ERROR: failed to add default include path /home/runner/work/FreeCAD-Bundle/FreeCAD-Bundle/conda/linux/AppDir/usr/share/X11/xkb
QOpenGLFunctions created with non-current context
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x19dc0) [0x7fb412245dc0]
#1  /tmp/.mount_FreeCAgDAIom/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_context_ref+0) [0x7fb3ffad5280]
#2  /tmp/.mount_FreeCAgDAIom/usr/plugins/platforms/../../lib/libxkbcommon.so.0(+0x1e775) [0x7fb3ffad7775]
#3  /tmp/.mount_FreeCAgDAIom/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_keymap_new_from_buffer+0x47) [0x7fb3ffad6f37]
#4  0x7fb40de17526 in QtWaylandClient::QWaylandInputDevice::Keyboard::keyboard_keymap(unsigned int, int, unsigned int) from /tmp/.mount_FreeCAgDAIom/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0xda
#5  /tmp/.mount_FreeCAgDAIom/usr/plugins/platforms/../../lib/./libffi.so.8(+0x6a4a) [0x7fb40dda7a4a]
#6  /tmp/.mount_FreeCAgDAIom/usr/plugins/platforms/../../lib/./libffi.so.8(+0x5fea) [0x7fb40dda6fea]
#7  /tmp/.mount_FreeCAgDAIom/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0xa573) [0x7fb410e2c573]
#8  /tmp/.mount_FreeCAgDAIom/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0x6950) [0x7fb410e28950]
#9  /tmp/.mount_FreeCAgDAIom/usr/plugins/platforms/../../lib/libwayland-client.so.0(wl_display_dispatch_queue_pending+0x6b) [0x7fb410e29a3b]
#10  0x7fb40de1b9b8 in QtWaylandClient::QWaylandDisplay::flushRequests() from /tmp/.mount_FreeCAgDAIom/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0x34
#11  0x7fb40de2171a in QtWaylandClient::QWaylandWindow::setVisible(bool) from /tmp/.mount_FreeCAgDAIom/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0x124
#12  0x7fb412d100d2 in QWindowPrivate::setVisible(bool) from /tmp/.mount_FreeCAgDAIom/usr/bin/../lib/libQt5Gui.so.5+0x1b8
#13  0x7fb41338ba7c in QWidgetPrivate::show_sys() from /tmp/.mount_FreeCAgDAIom/usr/bin/../lib/libQt5Widgets.so.5+0x2aa
#14  0x7fb413391847 in QWidgetPrivate::show_helper() from /tmp/.mount_FreeCAgDAIom/usr/bin/../lib/libQt5Widgets.so.5+0x123
#15  0x7fb413393672 in QWidgetPrivate::setVisible(bool) from /tmp/.mount_FreeCAgDAIom/usr/bin/../lib/libQt5Widgets.so.5+0x210
#16  0x7fb4154550a1 in Gui::MainWindow::startSplasher() from /tmp/.mount_FreeCAgDAIom/usr/bin/../lib/libFreeCADGui.so+0x2a1
#17  0x7fb414f9a592 in Gui::StartupPostProcess::showMainWindow() from /tmp/.mount_FreeCAgDAIom/usr/bin/../lib/libFreeCADGui.so+0x142
#18  0x7fb414f9c58d in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCAgDAIom/usr/bin/../lib/libFreeCADGui.so+0x6d
#19  0x7fb414ec8092 in Gui::Application::runApplication() from /tmp/.mount_FreeCAgDAIom/usr/bin/../lib/libFreeCADGui.so+0x4e2
#20  /tmp/.mount_FreeCAgDAIom/usr/bin/freecad(+0x78b7) [0x5568c80798b7]
#21  /lib64/libc.so.6(+0x3248) [0x7fb41222f248]
#22  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7fb41222f30b]
#23  /tmp/.mount_FreeCAgDAIom/usr/bin/freecad(+0x7c2a) [0x5568c8079c2a]
```

> Operating System: Fedora Linux 41
KDE Plasma Version: 6.2.2
KDE Frameworks Version: 6.7.0
Qt Version: 6.8.0
Kernel Version: 6.11.5-300.fc41.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 4 × Intel® Core™ i7-5600U CPU @ 2.60GHz
Graphics Processor: Mesa Intel® HD Graphics 5500

jmeurin, 2024-11-05:
FWIW, I see the same stack than kinghat on CrOS 132.0.6807.0 (Official Build) dev (64-bit).

adrianinsaval, 2024-11-09:
reverted wayland due to https://github.com/FreeCAD/FreeCAD/issues/17491#issuecomment-2442011253, but there are other problems with the appimages