FreeCAD/FreeCAD-Bundle #329: [Problem] Mac build is corrupt (Nov. 3)

Stokestack, 2024-11-03:
**Describe the bug**
Mac OS displays a dialog saying that FreeCAD is "damaged" upon launch.

**To Reproduce**
Steps to reproduce the behavior:
Download and install https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-39100-conda-macOS-arm64-py311.dmg

Launch FreeCAD from its icon or by double-clicking on an existing file.

**Screenshots**
Attached.

**Additional context**
I re-downloaded the DMG and got the same result.
[image]

oursland, 2024-11-03:
@Stokestack The weekly builds are not signed.  To run them you will need to run `xattr -c /Applications/FreeCAD.app` to remove the quarantine attribute associated with applications downloaded from the internet.

Stokestack, 2024-11-03:
Thanks, but that's not the reported problem. Unsigned apps result in a "This application can't be checked for..." or words to that effect.

oursland, 2024-11-04:
The behavior changed in macOS 15.  Universal and Apple Silicon applications no longer permit running unsigned with the quarantine attribute.  Unsigned Intel applications will provide the popup you're more familiar with.

Stokestack, 2024-11-04:
Interesting. I've seen the "can't be checked" message under OS 15. I'll file a bug report with Apple, because this behavior is wrong.

nicolasgalvez, 2024-11-04:
> @Stokestack The weekly builds are not signed. To run them you will need to run `xattr -c /Applications/FreeCAD.app` to remove the quarantine attribute associated with applications downloaded from the internet.

Thanks! Works for me.

> Interesting. I've seen the "can't be checked" message under OS 15. I'll file a bug report with Apple, because this behavior is wrong.

Thanks for filing a report, LMK if you want me to +1 it or whatever. That's weird behavior.

Stokestack, 2024-11-04:
Filed report FB15674712 with Apple, entitled _Unsigned applications are erroneously reported as "damaged."_

I will say, though that there is [something wrong with the current weekly build](https://github.com/FreeCAD/FreeCAD/issues/17658); although I doubt the message has anything to do with that.

adrianinsaval, 2024-11-13:
duplicate of #291

closed 2024-11-13 by adrianinsaval