FreeCAD/FreeCAD-Bundle #330: [Problem] Windows weekly not building

maxwxyz, 2024-11-05:
@adrianinsaval the Windows weekly build failed. Also manually starting the job resulted in an error.

oursland, 2024-11-05:
This appears to be an issue due to incompatible dependencies.  It should be resolved through the use of lock files.

Such lock file management is automated through `pixi`, but the current development of `pixi` using the same `conda-forge` dependencies results in a strange build failure on Windows that neither @looooo nor I understand: https://github.com/oursland/FreeCAD/actions/runs/11656954868/job/32453798456

maxwxyz, 2024-11-09:
Seems to work again

closed 2024-11-09 by maxwxyz

adrianinsaval, 2024-11-09:
I had to remove opencv so it could solve dependencies. It is not used in freecad itself so I think that's fine. I know assembly 4 used it but I think that doesn't work in 1.0 anyways