FreeCAD/FreeCAD-Bundle #331: [Problem] Launching an AppImage using the AppImageLauncher by TheAssassin

SpeeQz1, 2024-11-05:
**Describe the bug**
AppImages of the weekly build for x86_64 launched by the [AppImageLauncher](https://github.com/TheAssassin/AppImageLauncher) can't be launched and simply do not open.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the AppImageLauncher
2. Run the AppImage by clicking "Run Once" on the pop up menu when double clicking it
3. Alternatively you can also run the AppImageLauncher inside the terminal to see any errors

**Screenshots**
[image]

**Additional context**
Error produced by attempting to run the AppImage from the terminal which opens the menu to click "Run Once".

```
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_shall_not_be_integrated : sqfs_open_image error: /home/user/Downloads/FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311.AppImage
AppImageLauncher error: appimage_shall_not_be_integrated() failed (returned -1)
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_is_terminal_app : sqfs_open_image error: /home/user/Downloads/FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311.AppImage
AppImageLauncher error: appimage_is_terminal_app() failed (returned -1)
execv error: No such file or directory
```

OS: Linux Mint 22 (Cinnamon)

oursland, 2024-11-05:
The AppImage was built using the `zstd` compression algorithm, which has been a part of AppImage standard for a while now.  This addresses some performance issues that was being reported by users.  It appears the AppImageLauncher has not been updated to support this compression format.

Is AppImageLauncher still developed and supported?  The most recent release is from 2020.

SpeeQz1, 2024-11-05:
I don't know if it's being kept up to date, I tried running the AppImage using Gear Lever and it seems to work fine.

oursland, 2024-11-05:
Right, we updated the images last week to use `zstd` compression to address performance issues experienced by users.

adrianinsaval, 2024-11-05:
I would report this to appimagelauncher, beyond that I would recommend using appimagelauncher only for the desktop integration functionality and launch appimages directly the rest of the time

JohnOCFII, 2024-11-05:
I confirm this behavior with recent FreeCAD weeklies and AppImageLauncher (on Arm Debian Linux (bookworm) as a VM for me).  It looks like AppImageLauncher **does** support `zstd.` but there is some chat in one of their issues about level of compression.  Also - not sure if that is even in their 2020 release, or if you need to try the "continuous" build releases, which I haven't tried.

https://github.com/AppImage/AppImageKit/releases

https://github.com/AppImage/AppImageKit/issues/478#issuecomment-1944449840

SpeeQz1, 2024-11-06:
> I would report this to appimagelauncher, beyond that I would recommend using appimagelauncher only for the desktop integration functionality and launch appimages directly the rest of the time

AppImageLauncher actually replaces the launching functionality and you have to delete it to be able to launch AppImages without using it. As shown in the bug even attempting to run the AppImage from a terminal, it would still launch it using AppImageLauncher.

> I confirm this behavior with recent FreeCAD weeklies and AppImageLauncher (on Arm Debian Linux (bookworm) as a VM for me). It looks like AppImageLauncher **does** support `zstd.` but there is some chat in one of their issues about level of compression. Also - not sure if that is even in their 2020 release, or if you need to try the "continuous" build releases, which I haven't tried.
> 
> https://github.com/AppImage/AppImageKit/releases
> 
> [AppImage/AppImageKit#478 (comment)](https://github.com/AppImage/AppImageKit/issues/478#issuecomment-1944449840)

I can confirm the continuous build release cannot launch the AppImage on Mint.

Projekt-43, 2024-11-06:
I confirm that the workaround of @SpeeQz1:

> ... delete it to be able to launch AppImages without using it.

did work for me on Manjaro `6.10.13-3` to get passed the `zstd` issue! :tada: :pray: 

But then is the #332 issue...

leoheck, 2024-11-06:
Something has changed. This latest weekly is failing to launch.
Something changed in the Freecad to break the AppImages. 

https://github.com/FreeCAD/FreeCAD-Bundle/issues/332#issuecomment-2460587140

oursland, 2024-11-07:
The issue is that AppImageLauncher is no longer maintained and does not support images built by recent versions of the [`appimagetool`](https://appimage.github.io/appimagetool/).

oursland, 2024-11-08:
@Thiago-Assis-T Your issue is different from this one and is related to #332 .  There appears to be an issue with either `readline` or the `termcap` files in the most recent weekly build.

Thiago-Assis-T, 2024-11-10:
ooh, i see, thanks!

adrianinsaval, 2024-11-10:
so it doesn't sounds like there's much for us to do about this issue, I guess people should stop using appimage launcher... To replace it's functionality maybe we could include a script in the appimage that copies over the desktop file and icons? I'll see what I can come up with next week

X-Ryl669, 2024-11-14:
> I confirm that the workaround of @SpeeQz1:
> 
> > ... delete it to be able to launch AppImages without using it.
> 
> did work for me on Manjaro `6.10.13-3` to get passed the `zstd` issue! 🎉 🙏
> 
> But then is the #332 issue...

It doesn't for me:
```
/var/tmp/pamac-build-me/freecad-weekly-appimage/PKGBUILD: line 26: ./freecad-1.0.0_39176.AppImage: cannot execute: required file not found
```

adrianinsaval, 2024-11-14:
have you restarted after removing appimagelauncher? do you have fuse3 installed?

longrackslabs, 2024-11-14:
I can confirm that removing AppImageLauncher from my Linux Mint and chmod +x the RC4 app image the RC4 loads and runs.  Bummer about AppImageLauncher, but good for RC4?

adrianinsaval, 2024-11-14:
Although you can see some relatively recent commits there's been no actual code changes for over a year

eblanshey, 2024-11-17:
> so it doesn't sounds like there's much for us to do about this issue, I guess people should stop using appimage launcher... To replace it's functionality maybe we could include a script in the appimage that copies over the desktop file and icons? I'll see what I can come up with next week

That would be great, @adrianinsaval. The main issue that AppImageLauncher solves for me is not being able to pin FreeCAD to my dock in Ubuntu or have it show up in the list of applications, due to lack of a desktop file. If the AppImage can solve that natively, then the launcher is not needed.

longrackslabs, 2024-11-17:
Yes that sounds like the case.   FYI, Gear Lever seems to do that same
thing as AppLauncher.

eblanshey, 2024-11-17:
Confirmed that Gear Level works as intended.