FreeCAD/FreeCAD-Bundle #332: FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311.AppImage ends with segfault, build 39024 no problem.

huubvanniekerk, 2024-11-06:
Hi,

Running Fedora 40, build 39127 ends for me in segfault:

```
Cannot read termcap database;
using dumb terminal settings.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x40d00) [0x7fd7c666dd00]
#1  /lib64/libc.so.6(__libc_free+0x25) [0x7fd7c66d54f5]
#2  /tmp/.mount_FreeCALNfKNE/usr/lib/python3.11/lib-dynload/readline.cpython-311-x86_64-linux-gnu.so(PyInit_readline+0x15e) [0x7fd7a436368e]
#3  /tmp/.mount_FreeCALNfKNE/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x7fd7c8acc1cc]
#4  /tmp/.mount_FreeCALNfKNE/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x7fd7c8a025fb]
#5  /tmp/.mount_FreeCALNfKNE/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7fd7c89ed6ab]
#6  /tmp/.mount_FreeCALNfKNE/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7fd7c8a0bc1f]
#7  /tmp/.mount_FreeCALNfKNE/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7fd7c89fb376]
#8  /tmp/.mount_FreeCALNfKNE/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7fd7c8a2a1af]
#9  /tmp/.mount_FreeCALNfKNE/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7fd7c8a299d7]
#10  /tmp/.mount_FreeCALNfKNE/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7fd7c8a3eb5e]
```

However, build 39024 (the previous one) gives me no issues.

huubvanniekerk, 2024-11-06:
BTW, I forgot to mention that the total nr of error lines is #111.

Projekt-43, 2024-11-06:
Hi here,

I can confirm the same issue on Manjaro `6.10.13-3` with weekly-build `39127`.

luzpaz, 2024-11-06:
`FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311.AppImage` crashing on Manjaro 24.1  
FreeCAD 1.1.0, Libs: 1.1.0devR39127 (Git)

<details><summary>crash log</summary>

```
FreeCAD 1.1.0, Libs: 1.1.0devR39127 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Cannot read termcap database;
using dumb terminal settings.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7f3e5fc6a1d0]
#1  /usr/lib/libc.so.6(__libc_free+0x25) [0x7f3e5fcd2575]
#2  /tmp/.mount_FreeCAopDaJm/usr/lib/python3.11/lib-dynload/readline.cpython-311-x86_64-linux-gnu.so(PyInit_readline+0x15e) [0x7f3e4815d68e]
#3  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x7f3e620cc1cc]
#4  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x7f3e620025fb]
#5  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f3e61fed6ab]
#6  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3e6200bc1f]
#7  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7f3e61ffb376]
#8  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f3e6202a1af]
#9  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f3e620299d7]
#10  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7f3e6203eb5e]
#11  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f3e61ff3d46]
#12  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f3e61fed6ab]
#13  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3e6200bc1f]
#14  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7f3e61fd5e24]
#15  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x7f3e61f02bc7]
#16  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x29fbad) [0x7f3e6209fbad]
#17  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f3e6209f5a4]
#18  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x2b610e) [0x7f3e620b610e]
#19  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f3e61ff3d46]
#20  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f3e61fed6ab]
#21  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3e6200bc1f]
#22  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7f3e61ffb376]
#23  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f3e6202a1af]
#24  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f3e620299d7]
#25  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7f3e6203eb5e]
#26  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f3e61ff3d46]
#27  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f3e61fed6ab]
#28  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3e6200bc1f]
#29  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7f3e61fd5e24]
#30  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x7f3e61f02bc7]
#31  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3e6200bc1f]
#32  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7f3e61fd5e24]
#33  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x350) [0x7f3e61fd5ab0]
#34  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15(FinishSignatureInitialization+0x14a) [0x7f3e6150cd6a]
#35  /tmp/.mount_FreeCAopDaJm/usr/lib/python3.11/site-packages/PySide2/QtCore.cpython-311-x86_64-linux-gnu.so(PyInit_QtCore+0x27f4) [0x7f3e323572f4]
#36  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x7f3e620cc1cc]
#37  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x7f3e620025fb]
#38  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f3e61fed6ab]
#39  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3e6200bc1f]
#40  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7f3e61ffb376]
#41  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f3e6202a1af]
#42  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f3e620299d7]
#43  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7f3e6203eb5e]
#44  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f3e61ff3d46]
#45  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f3e61fed6ab]
#46  /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f3e6200bc1f]
[... cut ...]
#106  0x7f3e6299c58d in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libFreeCADGui.so+0x6d
#107  0x7f3e628c8092 in Gui::Application::runApplication() from /tmp/.mount_FreeCAopDaJm/usr/bin/../lib/libFreeCADGui.so+0x4e2
#108  /tmp/.mount_FreeCAopDaJm/usr/bin/freecad(+0x78b7) [0x564da05cb8b7]
#109  /usr/lib/libc.so.6(+0x25e08) [0x7f3e5fc52e08]
#110  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7f3e5fc52ecc]
#111  /tmp/.mount_FreeCAopDaJm/usr/bin/freecad(+0x7c2a) [0x564da05cbc2a]
```
</details>

leoheck, 2024-11-06:
Trying to run this `FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311_6dd31c684a359e531673fc582c662d3e.AppImage` but it looks like it its broken.
[image]

leoheck, 2024-11-06:
Is it possible to trigger a new build or something?

oursland, 2024-11-07:
@leoheck Your issue is different.  Line 1 of the output states the issue that AppImageLauncher does not support zstd compression: `Squashfs image uses (null) compression, this version supports only xz, zlib.`

oursland, 2024-11-07:
These line from `readline` seems a bit odd:

```
Cannot read termcap database;
using dumb terminal settings.
```

Is this using the host termcap db or is it expecting a termcap db in the AppImage?

mbramblett, 2024-11-07:
Same symptom using Mageia 9. From my Mageia applications menu I select the weekly build. A quick logo/image comes up and disappears.  A previous build - builds-39057 - runs with no issues...

[mike@localhost opt]$ ./FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311.AppImage --verbose
FreeCAD 1.1.0, Libs: 1.1.0devR39127 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Cannot read termcap database;
using dumb terminal settings.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x369e0) [0x7f37ae4649e0]
#1  /lib64/libc.so.6(__libc_free+0x1c) [0x7f37ae4c03ec]
#2  /tmp/.mount_FreeCAdgCGGA/usr/lib/python3.11/lib-dynload/readline.cpython-311-x86_64-linux-gnu.so(PyInit_readline+0x15e) [0x7f379825268e]
#3  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x7f37afacc1cc]
#4  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x7f37afa025fb]
#5  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f37af9ed6ab]
#6  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f37afa0bc1f]
#7  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7f37af9fb376]
#8  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f37afa2a1af]
#9  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f37afa299d7]
#10  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7f37afa3eb5e]
#11  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f37af9f3d46]
#12  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f37af9ed6ab]
#13  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f37afa0bc1f]
#14  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7f37af9d5e24]
#15  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x7f37af902bc7]
#16  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x29fbad) [0x7f37afa9fbad]
#17  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f37afa9f5a4]
#18  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x2b610e) [0x7f37afab610e]
#19  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f37af9f3d46]
#20  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f37af9ed6ab]
#21  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f37afa0bc1f]
#22  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7f37af9fb376]
#23  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f37afa2a1af]
#24  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f37afa299d7]
#25  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7f37afa3eb5e]
#26  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f37af9f3d46]
#27  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f37af9ed6ab]
#28  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f37afa0bc1f]
#29  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7f37af9d5e24]
#30  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x7f37af902bc7]
#31  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f37afa0bc1f]
#32  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7f37af9d5e24]
#33  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x350) [0x7f37af9d5ab0]
#34  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15(FinishSignatureInitialization+0x14a) [0x7f37aef42d6a]
#35  /tmp/.mount_FreeCAdgCGGA/usr/lib/python3.11/site-packages/PySide2/QtCore.cpython-311-x86_64-linux-gnu.so(PyInit_QtCore+0x27f4) [0x7f37807572f4]
#36  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x7f37afacc1cc]
#37  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x7f37afa025fb]
#38  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f37af9ed6ab]
#39  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f37afa0bc1f]
#40  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7f37af9fb376]
#41  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7f37afa2a1af]
#42  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7f37afa299d7]
#43  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7f37afa3eb5e]
#44  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7f37af9f3d46]
#45  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7f37af9ed6ab]
#46  /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7f37afa0bc1f]
[... cut ...]
#106  0x7f37b039c58d in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libFreeCADGui.so+0x6d
#107  0x7f37b02c8092 in Gui::Application::runApplication() from /tmp/.mount_FreeCAdgCGGA/usr/bin/../lib/libFreeCADGui.so+0x4e2
#108  /tmp/.mount_FreeCAdgCGGA/usr/bin/freecad(+0x78b7) [0x55e3ec4128b7]
#109  /lib64/libc.so.6(+0x23737) [0x7f37ae451737]
#110  /lib64/libc.so.6(__libc_start_main+0x85) [0x7f37ae4517f5]
#111  /tmp/.mount_FreeCAdgCGGA/usr/bin/freecad(+0x7c2a) [0x55e3ec412c2a]

JohnOCFII, 2024-11-07:
Duplicate of: https://github.com/FreeCAD/FreeCAD-Bundle/issues/331#issuecomment-2461045704

rgloor, 2024-11-08:
> Hi,
>
> Running Fedora 40, build 39127 ends for me in segfault:
>
> ```
> Cannot read termcap database;
> using dumb terminal settings.
> Program received signal SIGSEGV, Segmentation fault.
> #0 [...]

Confirm - same issue on openSUSE 15.6 (fully patched). 
weekly built 39092 running fine.

weekly build 39127 
and
weekly-buildd 39141
not working.

NB: python311-pyzstd  and python311-zstd are installed.

oursland, 2024-11-08:
This issue has been reproduced on Fedora 40, Manjaro, openSUSE, and Arch and does not appear to be distribution specific.  Given the error, I suspect an incompatible version of `readline` may have been installed during the building of the most recent weekly builds.

oursland, 2024-11-08:
There were a lot of changes in dependency versions between the last working build and this build:

|Package|Working | Broken |
|--------|---------|--------|
|blosc|1.21.5|1.21.6|
|ffmpeg|6.1.1|7.1.0|
|fontconfig|2.14.2|2.15.0|
|glib|2.80.2|2.82.2|
|glib-tools|2.80.2|2.82.2|
|gmsh|4.12.2|4.13.1|
|graphviz|11.0.0|12.0.0|
|gst-plugins-base|1.22.9|1.24.7|
|gstreamer|1.22.9|1.24.7|
|harfbuzz|8.5.0|9.0.0|
|icu|73.2|75.1|
|ifcopenshell|v0.7.0.231127|0.7.11|
|imath|3.1.11|3.1.12|
|jsoncpp|1.9.5|1.9.6|
|libabseil|20240116.2|20240722.0|
|libass|0.17.1|0.17.3|
|libboost|1.82.0|1.84.0|
|libboost-devel|1.82.0|1.84.0|
|libboost-headers|1.82.0|1.84.0|
|libclang|15.0.7|19.1.3|
|libclang13|15.0.7|19.1.3|
|libcurl|8.8.0|8.10.1|
|libdeflate|1.20|1.22|
|libglib|2.80.2|1.7.0|
|libhwloc|2.11.1|2.11.2|
|libllvm15|15.0.7|19.1.3|
|libnghttp2|1.58.0|1.64.0|
|libopenvino|2024.1.0|1.7.0|
|libopenvino-auto-batch-plugin|2024.1.0|2024.4.0|
|libopenvino-auto-plugin|2024.1.0|2024.4.0|
|libopenvino-hetero-plugin|2024.1.0|2024.4.0|
|libopenvino-intel-cpu-plugin|2024.1.0|2024.4.0|
|libopenvino-intel-gpu-plugin|2024.1.0|2024.4.0|
|libopenvino-intel-npu-plugin|2024.1.0|2024.4.0|
|libopenvino-ir-frontend|2024.1.0|2024.4.0|
|libopenvino-onnx-frontend|2024.1.0|2024.4.0|
|libopenvino-paddle-frontend|2024.1.0|2024.4.0|
|libopenvino-pytorch-frontend|2024.1.0|2024.4.0|
|libopenvino-tensorflow-frontend|2024.1.0|2024.4.0|
|libopenvino-tensorflow-lite-frontend|2024.1.0|2024.4.0|
|libpng|1.6.43|1.6.44|
|libprotobuf|4.25.3|5.27.5|
|libraw|0.21.1|0.21.3|
|librsvg|2.58.1|2.58.4|
|libsqlite|3.46.0|3.47.0|
|libtiff|4.6.0|4.7.0|
|libva|2.21.0|2.22.0|
|libxcb|1.15|1.17.0|
|libxml2|2.12.7|2.13.4|
|libzip|1.10.1|1.11.2|
|libzlib|1.2.13|1.3.1|
|llvm-openmp|19.1.2|19.1.3|
|lxml|5.2.2|5.3.0|
|mpg123|1.32.6|1.32.9|
|mysql-common|8.0.33|9.0.1|
|mysql-libs|8.0.33|9.0.1|
|nss|3.100|3.106|
|numpy|2.1.2|2.1.3|
|occt|7.7.2|7.8.1|
|pcl|1.14.0|1.14.1|
|pcre2|10.43|10.44|
|pip|24.2|24.3.1|
|proj|9.3.1|9.5.0|
|pulseaudio-client|16.1|17.0|
|pyside2|5.15.8|5.15.15|
|python|3.11.9|3.11.10|
|qt-main|5.15.13|5.15.15|
|setuptools|75.1.0|75.3.0|
|sqlite|3.46.0|3.47.0|
|svt-av1|2.1.0|2.3.0|
|tbb|2021.13.0|2022.0.0|
|tbb-devel|2021.13.0|2022.0.0|
|utfcpp|4.0.5|4.0.6|
|vtk|9.2.6|9.3.0|
|vtk-base|9.2.6|9.3.0|
|vtk-io-ffmpeg|9.2.6|9.3.0|
|xcb-util|0.4.0|0.4.1|
|xcb-util-keysyms|0.4.0|0.4.1|
|xcb-util-renderutil|0.3.9|0.3.10|
|xcb-util-wm|0.4.1|0.4.2|
|xkeyboard-config|2.42|2.43|
|xorg-libx11|1.8.9|1.8.10|
|xorg-libxext|1.3.4|1.3.6|
|xorg-libxfixes|5.0.3|6.0.1|
|xorg-libxi|1.7.10|1.8.2|
|xorg-libxmu|1.1.3|1.2.1|
|xorg-renderproto|0.11.1|1.1.5|
|xorg-xproto|7.0.31|2024.1|
|zlib|1.2.13|1.3.1|

mbramblett, 2024-11-08:
I have the same issue with FreeCAD_weekly-builds-39141-conda-Linux-x86_64-py311.AppImage

Thiago-Assis-T, 2024-11-09:
Hello, i've been having the same issue with the aur package, and out of frustration, i tried to open multiple times, and it actually worked!

if i open and wait, nothing happens, if i open like 3 or 4 times in a row, it opens, sometimes one instance and other times two or more...

hope this helps to diagnose the issue or people that need to run the software.

adrianinsaval, 2024-11-09:
what aur package are you using?

leoheck, 2024-11-09:
> Trying to run this `FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311_6dd31c684a359e531673fc582c662d3e.AppImage` but it looks like it its broken. [...]

@adrianinsaval the appimage works for me. I had to uninstall the AppImageLauncher (sick) and then I could use the AppImageLauncher.

Thiago-Assis-T, 2024-11-09:
> what aur package are you using? if you are using `freecad-weekly-appimage` then that's irrelevant since it is just the appimage that we publish, if you are using `freecad-git` then that would imply this is possibly a code issue and not a packaging issue

i am using the weekly appimage indeed

luzpaz, 2024-11-09:
`FreeCAD_weekly-builds-39141-conda-Linux-x86_64-py311.AppImage` crashing still. 

<details><summary>Segfault log</summary>

```shell
./FreeCAD_weekly-builds-39141-conda-Linux-x86_64-py311.AppImage                                                                                                                          1 ✘ 
FreeCAD 1.1.0, Libs: 1.1.0devR39141 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Cannot read termcap database;
using dumb terminal settings.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7ff9d566a1d0]
#1  /usr/lib/libc.so.6(__libc_free+0x25) [0x7ff9d56d2575]
#2  /tmp/.mount_FreeCAgNoOOM/usr/lib/python3.11/lib-dynload/readline.cpython-311-x86_64-linux-gnu.so(PyInit_readline+0x15e) [0x7ff9b9b7c68e]
#3  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x7ff9d7acc1cc]
#4  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x7ff9d7a025fb]
#5  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7ff9d79ed6ab]
#6  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ff9d7a0bc1f]
#7  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7ff9d79fb376]
#8  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7ff9d7a2a1af]
#9  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7ff9d7a299d7]
#10  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7ff9d7a3eb5e]
#11  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7ff9d79f3d46]
#12  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7ff9d79ed6ab]
#13  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ff9d7a0bc1f]
#14  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7ff9d79d5e24]
#15  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x7ff9d7902bc7]
#16  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x29fbad) [0x7ff9d7a9fbad]
#17  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7ff9d7a9f5a4]
#18  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x2b610e) [0x7ff9d7ab610e]
#19  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7ff9d79f3d46]
#20  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7ff9d79ed6ab]
#21  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ff9d7a0bc1f]
#22  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7ff9d79fb376]
#23  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7ff9d7a2a1af]
#24  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7ff9d7a299d7]
#25  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7ff9d7a3eb5e]
#26  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7ff9d79f3d46]
#27  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7ff9d79ed6ab]
#28  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ff9d7a0bc1f]
#29  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7ff9d79d5e24]
#30  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x102bc7) [0x7ff9d7902bc7]
#31  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ff9d7a0bc1f]
#32  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x1d5e24) [0x7ff9d79d5e24]
#33  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallFunction+0x350) [0x7ff9d79d5ab0]
#34  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15(FinishSignatureInitialization+0x14a) [0x7ff9d76c1d6a]
#35  /tmp/.mount_FreeCAgNoOOM/usr/lib/python3.11/site-packages/PySide2/QtCore.cpython-311-x86_64-linux-gnu.so(PyInit_QtCore+0x27f4) [0x7ff99fd572f4]
#36  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x2cc1cc) [0x7ff9d7acc1cc]
#37  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x2025fb) [0x7ff9d7a025fb]
#38  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7ff9d79ed6ab]
#39  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x17f) [0x7ff9d7a0bc1f]
#40  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x1fb376) [0x7ff9d79fb376]
#41  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(PyObject_CallMethodObjArgs+0x11f) [0x7ff9d7a2a1af]
#42  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(PyImport_ImportModuleLevelObject+0x787) [0x7ff9d7a299d7]
#43  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x23eb5e) [0x7ff9d7a3eb5e]
#44  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(+0x1f3d46) [0x7ff9d79f3d46]
#45  /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x81ab) [0x7ff9d79ed6ab]
[... cut ...]
#106  0x7ff9d839c58d in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libFreeCADGui.so+0x6d
#107  0x7ff9d82c8092 in Gui::Application::runApplication() from /tmp/.mount_FreeCAgNoOOM/usr/bin/../lib/libFreeCADGui.so+0x4e2
#108  /tmp/.mount_FreeCAgNoOOM/usr/bin/freecad(+0x78b7) [0x55bef1cac8b7]
#109  /usr/lib/libc.so.6(+0x25e08) [0x7ff9d5652e08]
#110  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7ff9d5652ecc]
#111  /tmp/.mount_FreeCAgNoOOM/usr/bin/freecad(+0x7c2a) [0x55bef1cacc2a]
```

</details>

adrianinsaval, 2024-11-10:
please test latest appimage, for me it now works on my fedora machine. I reverted to the same dependencies we where using before. I imagine something wen wrong in the qt5.15.15 upgrade, I'll see if I can fix it in the next few days

byteofsoren, 2024-11-10:
@adrianinsaval  New weekly AppImage (build 39153) works for me. Thanks.

mbramblett, 2024-11-10:
> @adrianinsaval New weekly AppImage (build 39153) works for me. Thanks.

Boots right up in Mandriva 9. Thanks!!!

ed3davis, 2024-11-10:
Good boot up now. Thank you.

OS: openSUSE Tumbleweed (KDE/plasmawayland/xcb)
Architecture: x86_64
Version: 1.1.0dev.39153 (Git) AppImage

luzpaz, 2024-11-10:
AppImage finally works again  
```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39153 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Silk 0.1.5
```

luzpaz, 2024-11-10:
@oursland how did you generate this list ? (https://github.com/FreeCAD/FreeCAD-Bundle/issues/332#issuecomment-2465519884)

oursland, 2024-11-11:
This was generated "manually" by copying the installed files from the logs of each build into a files.  I sorted the files alphabetically and ran `diff -y --minimal [log1] [log2]`.  Then only retained the results that contained the `>` character.

huubvanniekerk, 2024-11-11:
I'm happy to see and report that build 39153 does work for me on Fedora 40. Thanks for all the hard work to fix it.

adrianinsaval, 2025-05-16:
can anybody that faced this issue before please test if it's also a thing on the freecad 1.0.1 release? https://github.com/FreeCAD/FreeCAD/releases/tag/1.0.1

I could not reproduce it on fedora 40 anymore but there is a risk that it's still a thing on other systems

hyarion, 2025-05-19:
> can anybody that faced this issue before please test if it's also a thing on the freecad 1.0.1 release? https://github.com/FreeCAD/FreeCAD/releases/tag/1.0.1
> 
> I could not reproduce it on fedora 40 anymore but there is a risk that it's still a thing on other systems

@huubvanniekerk can you check it?