FreeCAD/FreeCAD-Bundle #334: FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311.AppImage fails to start, FUSE access denied

DerMouse, 2024-11-07:
Gentoo Linux here.

The latest weekly triggers a FUSE error : 

```
$ ./FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311.AppImage 
fuse: failed to exec fusermount: Permission denied

Cannot mount AppImage, please check your FUSE setup.
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
open dir error: No such file or directory
```
Previous weeklies (39057 and 39092) run without any kind of problem.

Maybe of interest : the 39127 weekly is statically linked, whereas the two others are not : 

```
$ file FreeCAD_weekly-builds-39*
FreeCAD_weekly-builds-39057-conda-Linux-x86_64-py311.AppImage: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=30e06184968532b6a9aa36f44ada39e4af0bda56, for GNU/Linux 2.6.32, stripped
FreeCAD_weekly-builds-39092-conda-Linux-x86_64-py311.AppImage: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=30e06184968532b6a9aa36f44ada39e4af0bda56, for GNU/Linux 2.6.32, stripped
FreeCAD_weekly-builds-39127-conda-Linux-x86_64-py311.AppImage: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), statically linked, BuildID[sha1]=9d7e67802fffc45391e9a993c33120227d1b7c78, stripped
```

If extracting the 39127 (run with --appimage-extract), the squashfs-root/usr/bin/freecad  executable runs without any problem either.

oursland, 2024-11-08:
We recently upgraded the compression to `zstd`, which is produced by the latest version of `appimagetool`.

Do you know more about your install's use of `fuse` and if it supports this compression format?

@adrianinsaval ping

DerMouse, 2024-11-08:
> We recently upgraded the compression to `zstd`, which is produced by the latest version of `appimagetool`.
> 
> Do you know more about your install's use of `fuse` and if it supports this compression format?

If I run the 39127 AppImage as root, everything works OK. So I looked up here and there, and found https://github.com/AppImage/type2-runtime/issues/15 issue which prettily summarizes the problem. It's somewhere in AppImage's code.

Ad hoc workaround, add (as root) read rights for "others" on the fusermount executable : chmod o+r /usr/bin/fusermount*

So not a FreeCad issue.

DerMouse, 2024-11-08:
Closing this issue.

DerMouse, 2024-11-08:
Closing the issue.

closed 2024-11-08 by DerMouse