FreeCAD/FreeCAD-Bundle #336: FreeCAD weekly build app image is broken

X-Ryl669, 2024-11-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Recently - last weekly build that's known to work is 39057, the AppImage generated in the weekly build changed its format somehow. Previously, one could do `./FreeCAD.appimage --appimage-extract org.freecad.FreeCAD.desktop` and it would extract this file.

However, since 39057, running the above command results in:
```
This doesn't look like a squashfs image.
Failed to open squashfs image
```

This is an issue for package managers (like on Arch linux) that are patching the desktop file to match their distributions' path (see [here](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=freecad-weekly-appimage))

This fails and breaks update for all other packages, so in the short term, this implies removing the package from the repositories, which is bad for FreeCAD diffusion.

Is there anything changed that would explain why the appimage's image has changed format?

### Full version info

```shell
39176
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-14:
@adrianinsaval FYI

adrianinsaval, 2024-11-14:
I can't reproduce this on arch with the latest appimage

X-Ryl669, 2024-11-14:
Got this when building this [package](https://aur.archlinux.org/packages/freecad-weekly-appimage):
```
Construction de freecad-weekly-appimage...
==> Création du paquet freecad-weekly-appimage 1.0.0_39176-1 (jeu. 14 nov. 2024 17:43:08)
==> Vérification des dépendances pour l’exécution…
==> Vérification des dépendances pour la compilation…
==> Récupération des sources…
  -> freecad-1.0.0_39176.AppImage trouvé
  -> freecad.sh trouvé
  -> org.freecad.FreeCAD.desktop.patch trouvé
==> Validation des fichiers source avec sha256sums…
    freecad-1.0.0_39176.AppImage ... Ignoré
    freecad.sh ... Réussite
    org.freecad.FreeCAD.desktop.patch ... Réussite
==> Suppression du répertoire $srcdir/ existant…
==> Extraction des sources…
==> Lancement de prepare()…
This doesn't look like a squashfs image.
Failed to open squashfs image
==> ERREUR : Une erreur s’est produite dans prepare().
    Abandon…
```
Looking at the pkgbuild, it's in the prepare and it fails at the step "--appimage-extract". So I downloaded the Appimage from the repository and tried to run the "--appimage-extract" command and get the exact same result: `This doesn't look like a squashfs image`, for me it's the appimage that's broken, it's not able to extract itself. 

 I can't run the appimage either:
```
$ ./FreeCAD_weekly-builds-39176-conda-Linux-x86_64-py311.AppImage
execv error: No such file or directory
```

The other appimage I've download today to test works fine (like OrcaSlicer).

adrianinsaval, 2024-11-14:
The AUR package built without issues for me. I'm not sure what's different for you but it might be a permissions problem. Can you try on any other system?

adrianinsaval, 2024-11-14:
Ok I figured it out, I installed appimagelauncher in my system and I see the same errors, appimagelauncher has not been updated in a while and does not support the newest appimage tooling, I recommend uninstalling appimagelauncher, see #331 and https://github.com/TheAssassin/AppImageLauncher/issues/602

closed 2024-11-14 by adrianinsaval