FreeCAD/FreeCAD-Bundle #337: RC4 Appimage can't find fusermount

pyro9, 2024-11-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Ubuntu 18.04.6 LTS

When running FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage it reports: 
fuse: failed to exec fusermount: Permission denied

Worked around with ln -s /bin/fusermount /usr/bin/fusermount

RC2 works and previous works without the workaround.

### Full version info

```shell
after workaround:
OS: Ubuntu 18.04.6 LTS (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
  * ThreadProfile 1.90.0
  * Curves 0.6.51
  * 3D_Printing_Tools
  * lattice2 1.0.0
  * Silk 1.0.0
  * Assembly4 0.50.15
  * A2plus 0.4.68
  * CfdOF 1.21.6
  * fcgear.backup1677967119.4973533 1.0.0 (Disabled)
  * fcgear 1.0.0
  * sheetmetal 0.2.61
  * Render 2023.8.18
  * CurvedShapes 1.0.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-11-18:
also happens in Ubuntu20

MichielBruijn, 2024-11-18:
Can't get RC4 to work in ubuntu 24.04.

michiel@clevo:~/Downloads$ ./FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage 
QSocketNotifier: Can only be used with threads started with QThread
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_shall_not_be_integrated : sqfs_open_image error: /home/michiel/Downloads/FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage
AppImageLauncher error: appimage_shall_not_be_integrated() failed (returned -1)
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_is_terminal_app : sqfs_open_image error: /home/michiel/Downloads/FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage
AppImageLauncher error: appimage_is_terminal_app() failed (returned -1)
execv error: No such file or directory

Possible the same error?

I've tried to let chatgpt solve the problem but none of the options worked.
Couldn't even extract the appimage

adrianinsaval, 2024-11-18:
> AppImageLauncher error: appimage_is_terminal_app() failed (returned -1) execv error: No such file or directory
> 
> Possible the same error?

not the same error, this is #331 uninstall appimagelauncher and restart your computer

MichielBruijn, 2024-11-18:
> > AppImageLauncher error: appimage_is_terminal_app() failed (returned -1) execv error: No such file or directory
> > Possible the same error?
> 
> not the same error, this is #331 uninstall appimagelauncher and restart your computer

Thanks, that was the problem indeed

adrianinsaval, 2024-11-19:
Please test the latest weekly or the 1.0 release appimage, I used a patched appimage runtime so it should work now. I tested on an Ubuntu 18.04 container and it worked

pyro9, 2024-11-19:
> Please test the latest weekly or the 1.0 release appimage, I used a patched appimage runtime so it should work now. I tested on an Ubuntu 18.04 container and it worked

That fixed it. I removed my workaround then started the 1.0 stable release without issue.

closed 2024-11-20 by maxwxyz