FreeCAD/FreeCAD-Bundle #338: x86-64 RC4 not running on archlinux

Fulgacian, 2024-11-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Running the appimage, it throws the error:
execv error: No such file or directory

### Full version info

```shell
FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage
```

### Subproject(s) affected?

None

### Anything else?

OS: ArcoLinux x86_64
Kernel: Linux 6.11.7-zen1-1-zen
DE: Xfce4 4.18

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-11-16:
Duplicate of #331 and #336, appimagelauncher is not compatible with latest appimages

mendberg, 2024-11-18:
I made the AppImage executeable and still get an error when I start it from terminal on **Manjaro** linux (/Gnome)

```
:  chmod +x FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage 
:  ./FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
execv error: No such file or directory
```

tarman3, 2024-11-18:
Did you try to use this commands to run?

`QT_QPA_PLATFORM=wayland ./FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage`
`QT_QPA_PLATFORM=xcb ./FreeCAD_1.0.0RC4-conda-Linux-x86_64-py311.AppImage`

mendberg, 2024-11-18:
Thanks. Those commands get corrected to "XWayland" and "dcb" – both fail.

tarman3, 2024-11-18:
Freecad 1.0rc4-3 package in testing stage of manjaro repository.
Perhaps the easiest thing to do would be to just wait a little.

mendberg, 2024-11-18:
I'm not hurrying. I just want to report issues so that potential bugs can be fixed in time.

mendberg, 2024-11-18:
No, of course not. I'm not gonna break my whole system because one single AppImage decided to make trouble.

oursland, 2024-11-18:
The issue is that AppImageLauncher is no longer maintained and does not support perfectly valid AppImages produced by the `appimagetool`.  FreeCAD may be the first of the incompatible AppImages that you have encountered, but there are others and will be more as newer releases are made with more recent versions of `appimagetool`.

tarman3, 2024-11-18:
> No, of course not. I'm not gonna break my whole system because one single AppImage decided to make trouble.

I download **manjaro-gnome-24.1.2-241104-linux610.iso** and run it in **qemu**.
**FreeCAD_weekly-builds-39215-conda-Linux-x86_64-py311.AppImage** run without problem.

adrianinsaval, 2024-11-18:
I will close this issue as a duplicate, the problem is in appimagelauncher

tarman3, 2024-11-18:
> I made the AppImage executeable and still get an error when I start it from terminal on **Manjaro** linux (/Gnome)

I download manjaro-gnome-24.1.2-241104-linux610.iso and run it in qemu.
FreeCAD_weekly-builds-39215-conda-Linux-x86_64-py311.AppImage run without problem.

tarman3, 2024-11-18:
> O, this is the first time I read about AppimageLauncher being discontinued. Source for those bad news? I'm not afraid of using the terminal. But there should be a more user friendly way to use AppImage…
You can create .desktop file for any AppImage file and run it like any other application

adrianinsaval, 2024-11-18:
I don't think it's officially discontinued, but it has not been updated for some time which is important because it doesn't support the newer appimage tooling which has been out for a while, many appimages are not working with it anymore.

tarman3, 2024-11-18:
> RC4 runs like in a dream

PS You can add more than one .desktop file for FreeCAD release and Dev version.

mendberg, 2024-11-18:
Did i say "runs like a dream"? Well, it must have been me who was dreaming… didn't test the SpaceMouse – which is no longer working w/ the AppImages since removing that launcher. :-( 
see https://github.com/FreeCAD/FreeCAD/issues/17956