FreeCAD/FreeCAD-Bundle #343: Tree View font sixe not persisted...

CIKoolK, 2024-11-25:
I have found that the Edit|Preferences|Display|Advanced Tree View font setting while dynamically effective, is not persisted between sessions:
1. Upon initial startup it is set and displayed as 10 pt.;
2. I adjust it to 13 pt., click Apply, and it is applied;
3. Shut down and restart, it is sshown as 13 pt. in the Preferences, but displayed in 10 pt.

[image]

[image]

[image]

As I have reported previously, I have significant vision problems making this sort of thing bothersome...

CIKoolK, 2024-12-07:
Still not working in build 39503...

Syres916, 2024-12-11:
@CIKoolK hopefully you will see the fix in the next week or two but in future please log application issues at https://github.com/FreeCAD/FreeCAD/issues this repository is only for the actual _packaging_ not the _code_.

CIKoolK, 2024-12-11:
Will do, thank you!

closed 2024-12-13 by chennes