FreeCAD/FreeCAD-Bundle #344: macOS arm64 "FreeCAD-1.1.0dev39315" is damaged and can't be opened. You should move it to the Trash.

winstonford, 2024-11-26:
**Describe the bug**
When attempting to open the app on macOS arm64, I get a dialogue "FreeCAD-1.1.0dev39315" is damaged and can't be opened. You should move it to the Trash."

I have gotten this same error on the last two weeklies: 39240, and 39215.

However, on macOS x86, I am able to open 1.1.0dev39315, 39240, and 39215 normally, with no problems.

**To Reproduce**
Steps to reproduce the behavior:
1. Download weekly
2. Double click .dmg
3. Copy app to /Applications
4. Double click on app, or right click open app.

**Screenshots**

[image]

**Additional context**

oursland, 2024-11-26:
The weekly builds are not signed.  When downloaded from your browser, a `quarantine` flag is added to the file.  To remove the `quarantine` run the following command:

`xattr -cr /Applications/FreeCAD.app`

winstonford, 2024-11-26:
Great, all opening normally.  Thanks for the tip.  I suppose the tip could be added to a readme file in the dmg.

closed 2024-11-26 by winstonford